StefanSchippers/xschem #1: Drawing issue

algorithm314, 2020-09-29:
Hello, at first thank you for your awesome tool.
I am a noob that starts to learn analog vlsi.

I have a weird issue about the rendering of schematics. I can not explain it easily.
Here is a [video](http://snf-869216.vm.okeanos.grnet.gr/xschem.mkv)
If i press ESC everything gets normal again.

I use xschem 2.9.7 in Debian testing.

Also it would be very helpful if there was an option that net names were displayed in pins like in Cadence.
Now i have to press right click, see the name and then close the window.

StefanSchippers, 2020-09-29:
Hi, thank you for for your feedback.
The rendering issue is a problem with some graphic cards. I have a radeon (AMD)  display adapter and i had the same issue,
i had to create this simple` /etc/X11/xorg.conf` file to solve the issue:

```
Section "Device"
        Identifier  "radeon"
        Option "ShadowPrimary" "true"
        # Following option can be used if you don't have a radeon/amd chipset just to verify if
        # problem is due to graphic accelerator. Of course not a solution for xschem as
        # it disables acceleration completely.
        # Option Accel "false"

EndSection
```
Try to create (as root) this file and restart the graphic session.
If the weird issue is gone then it is a problem with the graphic card/Xserver not correctly implementing the Xorg/Xlib specifications.

From the driver manual:

```
Option "ShadowPrimary" "boolean"
This option enables a so-called "shadow primary" buffer for fast CPU access to pixel data,
and separate scanout buffers for each display  controller  (CRTC).
This may improve performance for some 2D workloads, potentially at the expense of other
(e.g. 3D, video) workloads.  Note in particular that enabling this option currently disables
page flipping.  The default is off.
```
For the net names the correct thing to do is to attach the 'devices/lab_wire.sym' component to a net and set the name (pressing 'q'):

[image]

algorithm314, 2020-09-29:
Thank you i found the issue, I needed to use Glamor acceleration (ShadowPrimary is not needed)

```
Section "Device"
        Identifier  "amdgpu"
        Option "AccelMethod" "Glamor"
EndSection
```

As for the names, yes you can insert pins to set names, but it is extra work if you want to quickly debug a circuit.

StefanSchippers, 2020-09-29:
> As for the names, yes you can insert pins to set names, but it is extra work if you want to quickly debug a circuit.

That is a good suggestion anyway. I am working on this. I guess you want a component display the net name it is connected to, like in this picture?
[image]

algorithm314, 2020-09-29:
Yes looks awesome

StefanSchippers, 2020-09-29:
I have implemented this feature in latest commit. details shown in this [Video](https://xschem.sourceforge.io/stefan/xschem_man/video_tutorials/net_name_attribute.mp4). @algorithm314, if that works for you i will close the issue. In the mean time I will do some testing.

algorithm314, 2020-09-30:
I was thinking for a single button or shortcut to do this, but its ok. You can close the issue.

StefanSchippers, 2020-09-30:
yes, adding this will be done asap... . :-) thank you!

StefanSchippers, 2020-10-02:
@algorithm314, the shortcut to enable/disable pin net name displaying is done. Check out the updated [Video](https://xschem.sourceforge.io/stefan/xschem_man/video_tutorials/net_name_attribute.mp4). Any feedback is welcome!.

closed 2020-10-02 by StefanSchippers