StefanSchippers/xschem #4: Generated PDF does not have searchable text

bluecmd, 2020-10-04:
Hi,

Exporting a PDF from xschem seems to produce a PDF that looks correct, but the text elements seems to be vectors and not searchable.

It would be very nice if you could search for component names etc in the generated PDF.

To reproduce:

1) `xschem /usr/local/share/doc/xschem/examples/rlc.sch`
2) File -> PDF export
3) Export as `rlc.pdf`
4) Open `rlc.pdf`
5) Try to search for e.g. `.save all`. No hits are found

StefanSchippers, 2020-10-07:
PDF is generated by converting a postscript file (usually using ps2pdf, the tool to use can be configured in xschemrc), and you are right, text is not generated using postscript fonts but using xshem's internal vector ('plotter style') font, so at the end there are only segents, no real text. Using Postscript fonts (this is what postscript was designed for)  is for sure a future improvement. Just need to learn how to correctly place, size and rotate text.

StefanSchippers, 2020-11-18:
@bluecmd the same problem used to be in svg exports. In latest commit svg exported images now use true fonts (using the svg <text> element) and text is thus searchable (excluding escaped XML characters <, >, &).
Since the results are really good the plan is to do the same for ps/pdf exports. This makes exported files also much smaller and faster to render.

StefanSchippers, 2020-12-16:
@bluecmd, after a rather long time i finally cleaned up also the postscript/pdf export.
now postscript (and the associated pdf converted file) use regular fonts so text is searchable, but more importantly of better quality. If you specify a .ps extension when doing a ps/pdf export you get plain postscript, while using .pdf you get pdf.
The xschem window aspect ration decides if a portrait layout or landscape layout is to be used.
here a snapshot of the pdf result on screen.
[image]

closed 2020-12-16 by StefanSchippers