StefanSchippers/xschem #5: set replace_key not working

chrische-xx, 2021-01-17:
Hi,
Thanks for creating this tool! I really like to use it, but one thing that is very important 
to me is that i can create custom keybindings. 

I found that although the xschem allows to remap keys the following addition to my xschemrc has no effect: 
```
set replace_key(d, Delete)
```
I also found that in xschem.tcl  set_replace_key_binding is called but it never enters the conditional if statement
```
proc set_replace_key_binding {} {
  if {[array exists replace_key]} {
    foreach i [array names replace_key] {
      key_binding "$i" "$replace_key($i)"
    }
  }
}
```
Therefore key_binding is never called. When manually applying a call from set_replace_key_binding I can force a remap that works in xschem. 

```

proc set_replace_key_binding {} {
  key_binding d Delete
  if {[array exists replace_key]} {
    foreach i [array names replace_key] {
      key_binding "$i" "$replace_key($i)"
    }
  }
}
```

StefanSchippers, 2021-01-17:
Thank you for pointing this out, will fix ASAP. the problem was a missing '**global replace_key**' inside the set_replace_key_binding procedure. This is a common error when taking some lines of tcl code and making a procedure :-)
```
proc set_replace_key_binding {} {
  global replace_key
  if {[array exists replace_key]} {
    foreach i [array names replace_key] {
      key_binding "$i" "$replace_key($i)"
    }
  }
}
```

StefanSchippers, 2021-01-17:
@chrische-xx  it seems fixed now.
The correct way to set keybindings is (in your `xschemrc` file): 
`set replace_key(Key-d)  Delete`
when the key has no modifiers (Shift/Control) its better to add `Key-`, i found that otherwise xschem may get confused, for example:
`set replace_key(1) w `
does not work, mat be because the '1' is interpreted as a numeric, don't know exactly.
doing:
`set replace_key(Key-1) Key-w`
does work as expected.
If modifiers are present (like: `Shift-W`) no 'Key-' is necessary.

closed 2021-01-17 by StefanSchippers

chrische-xx, 2021-01-17:
Works great now, thanks for the quick response!