StefanSchippers/xschem #6: Edit netlist not opening

mdzakirhussain, 2021-02-09:
Kindly help me. In my case, after clicking on netlist and then going to edit netlist, there's no netlist opening. Please help.

StefanSchippers, 2021-02-10:
Hi, do these checks.

1. After pressing 'Netlist' try to see if there is a .spice file in your home directory under .xschem/simulations.
   For example if your circuit name is test.sch there should be a ~/.xschem/simulations/test.spice
2. to display the netlist xschem uses an editor. Default is gvim (the command that is run in the background is gvim -f test.spice). You can edit your xschemrc file and add a different editor:
   set editor xedit

So if in item 1. you see the .spice file the problem is only displaying the file, just fix item 2.

let me know if the problem persists. Thank you for reporting.

mdzakirhussain, 2021-02-13:
@StefanSchippers Sir, thank you so much it worked.

closed 2021-02-13 by mdzakirhussain