StefanSchippers/xschem #7: While simulation ngspice reports models not found

mdzakirhussain, 2021-02-13:
Sir, I had tried inverter by picking NMOS and PMOS from devices. And had given vdd and vin for the inverter. While I try to simulate ngspice reports errors , such as NMOS and PMOS models not found. 

How do I add any model file in xschem ? to avoid the aforesaid problem.

StefanSchippers, 2021-02-13:
Hi, i can not supply cmos models with xschem as most of these are non free.
You can download pmos and nmos models here:
http://ptm.asu.edu/modelcard/130nm_bulk.txt
this gives you a reasonable 130nm process model for nmos and pmos
save this into a file 'models.txt' and add a '.include models.txt' into your spice commands.

mdzakirhussain, 2021-02-14:
@StefanSchippers sir, after adding 130nm_bulk I am getting this error 
[image]

StefanSchippers, 2021-02-14:
Use these models
[models_cmos_example.txt](https://github.com/StefanSchippers/xschem/files/5978404/models_cmos_example.txt)

mdzakirhussain, 2021-02-17:
@StefanSchippers sir, Thank you so much the file you gave the second time had solved the problem. 

Few Questions(out of my ignorance)
1. Why had the first time the given file failed? 
2. Is it possible to use sky130 instead of the 2nd time given file? If yes, then how do we do the use of sky130 models?

StefanSchippers, 2021-02-17:
Hi, 
The first model file is working only for a channel lengh=130nm, so L=130nm for all transistors. So it is basically my fault, this file is not good for generic design and simulation.
For xschem + sky130 there is a [video ](https://xschem.sourceforge.io/stefan/xschem_man/video_tutorials/install_xschem_sky130_and_ngspice.mp4), if you can not see it there is a [youtube ](https://www.youtube.com/watch?v=jXmmxO8WG8s) version. 
You need to clone this repository for the [xschem sky130](https://github.com/StefanSchippers/xschem_sky130) symbols and libraries. You also need to download the skywater-pdk. Refer to the video for instructions.

mdzakirhussain, 2021-02-25:
@StefanSchippers Sir, I tried all that. But, I want is this 

The circuit for example simple inverter it must be making use of sky130 models for its PMOS and NMOS. I am not able to do that. 
what I am able to be is, am making an inverter by using PMOS and NMOS but when it comes to simulation, I must be letting the tool know where are those models of PMOS and NMOS. I am not able to do that, please help how would I do that?

StefanSchippers, 2021-02-25:
@mdzakirhussain after installling xschem you have to install [xschem_sky130](https://github.com/StefanSchippers/xschem_sky130) libraries, after installing go into the xschem_sky130/ directory, run xschem from that directory, 
the schematic contains lot of examples both analog and digital, that use sky130 models.

mdzakirhussain, 2021-02-26:
@StefanSchippers sir, 
Yes sir the way you had instructed I could able to do that. But, my question is..... when I make my own inverter using PMOS and NMOS devices **in the new schematic window**, I must instruct the tool to make use of sky130 PMOS and sky130 NMOS models right..... he where I am facing difficulty how do I do that. That is, how do I add sky130 PMOS and sky130 NMOS models in the new schematic wherein I have my CMOS inverter(which is obviously made up of PMOS and NMOS devices).

mdzakirhussain, 2021-02-26:
@StefanSchippers sir,
What I am looking for is, I want to invoke/launch/run tool from any directory and want to design my own circuit using PMOS and NMOS devices, **and wanted to instruct the tool to use sky130 models**. How do I do that? is the question sir?

StefanSchippers, 2021-02-26:
in the example schematics there is a "models" component. Copy it (ctrl-c)
and paste it into your schematic( ctrl-v)

mdzakirhussain, 2021-02-27:
@StefanSchippers sir,
I did the way you had instructed but i am getting this 
[image]

mdzakirhussain, 2021-02-27:
@StefanSchippers sir, 
I humbly request you to share a video wherein simple inverter circuit is made using sky130 models.

StefanSchippers, 2021-02-27:
Try this tutorial by @bluecmd

https://github.com/bluecmd/learn-sky130/blob/main/schematic/xschem/getting-started.md

StefanSchippers, 2021-03-07:
Can you attach here your inv.sch file, your xschemrc file and the generated inv.spice netlist?

mdzakirhussain, 2021-03-09:
@StefanSchippers sir,

First of all, I am sorry for responding lately. Thank you so much sir for your direction to the tutorial. After completing the schematic I just have added from the tutorial the following and it has worked flawlessly.

name=SPICE only_toplevel=false value=".lib /usr/local/share/sky130_fd_pr/models/sky130.lib.spice tt
.tran 0.1n 1u
.save all"

closed 2021-03-17 by StefanSchippers