StefanSchippers/xschem #8: xpm in installed but while configuring it says not found.

ombhilare999, 2021-03-06:
Hi @StefanSchippers, I was trying to install xschem.
Following the manual on xschem site:
> http://repo.hu/projects/xschem/xschem_man/install_xschem.html
But I got this error, kindly help
Thank you
[image]

ombhilare999, 2021-03-06:
I figured out the issue
libxpm was missing.
Thank you

closed 2021-03-06 by ombhilare999

StefanSchippers, 2021-03-06:
Thank you for closing the issue, i came here too late, but happy you were able to get xschem build.
Yes, libxpm (or libxpm4 on some systems) as well as libxpm-dev (for header files) is listed in the depenciences on the 'install xschem' page.
Regards,

Magaspa, 2022-09-15:
> Hi @StefanSchippers, I was trying to install xschem. Following the manual on xschem site:
> 
> > http://repo.hu/projects/xschem/xschem_man/install_xschem.html
> > But I got this error, kindly help
> > Thank you
> > [image]

Can you tell me how to fix it?