StefanSchippers/xschem #9: Compiling Error

abhisupa, 2021-03-07:
abhishek@abhishek:~$ git clone https://github.com/StefanSchippers/xschem.git xschem_git
Cloning into 'xschem_git'...
remote: Enumerating objects: 14, done.
remote: Counting objects: 100% (14/14), done.
remote: Compressing objects: 100% (12/12), done.
remote: Total 6724 (delta 2), reused 6 (delta 2), pack-reused 6710
Receiving objects: 100% (6724/6724), 13.60 MiB | 10.60 MiB/s, done.
Resolving deltas: 100% (4250/4250), done.
abhishek@abhishek:~$ cd xschem_git/
abhishek@abhishek:~/xschem_git$ ls
AUTHORS    config.h.in  COPYING  INSTALL  Makefile          README           README.md  src    xschem_library
Changelog  configure    doc      LICENSE  Makefile.conf.in  README_MacOS.md  scconfig   tests  XSchemWin
abhishek@abhishek:~/xschem_git$ ./configure --prefix=/home/schippes
cc -DGENCALL -DRUNTIME -g  -DPLUGIN_SCRIPTS   -Isrc//tmpasm  -DPLUGIN_PARSGEN    -DPLUGIN_GUI     -Isrc//default   -c -o hooks.o hooks.c
cc -DGENCALL -DRUNTIME -g  -DPLUGIN_SCRIPTS   -Isrc//tmpasm  -DPLUGIN_PARSGEN    -DPLUGIN_GUI     -Isrc//default -c src//default/find_cc.c -o src//default/find_cc.o
cc -DGENCALL -DRUNTIME -g  -DPLUGIN_SCRIPTS   -Isrc//tmpasm  -DPLUGIN_PARSGEN    -DPLUGIN_GUI     -Isrc//default -c src//default/lib_compile.c -o src//default/lib_compile.o
cc -DGENCALL -DRUNTIME -g  -DPLUGIN_SCRIPTS   -Isrc//tmpasm  -DPLUGIN_PARSGEN    -DPLUGIN_GUI     -Isrc//default -c src//default/lib_uniqinc.c -o src//default/lib_uniqinc.o
cc -DGENCALL -DRUNTIME -g  -DPLUGIN_SCRIPTS   -Isrc//tmpasm  -DPLUGIN_PARSGEN    -DPLUGIN_GUI     -Isrc//default -c src//default/lib_file.c -o src//default/lib_file.o
cc -DGENCALL -DRUNTIME -g  -DPLUGIN_SCRIPTS   -Isrc//tmpasm  -DPLUGIN_PARSGEN    -DPLUGIN_GUI     -Isrc//default -c src//default/lib_try.c -o src//default/lib_try.o
cc -DGENCALL -DRUNTIME -g  -DPLUGIN_SCRIPTS   -Isrc//tmpasm  -DPLUGIN_PARSGEN    -DPLUGIN_GUI     -Isrc//default -c src//default/str.c -o src//default/str.o
cc -DGENCALL -DRUNTIME -g  -DPLUGIN_SCRIPTS   -Isrc//tmpasm  -DPLUGIN_PARSGEN    -DPLUGIN_GUI     -Isrc//default -c src//default/ht.c -o src//default/ht.o
cc -DGENCALL -DRUNTIME -g  -DPLUGIN_SCRIPTS   -Isrc//tmpasm  -DPLUGIN_PARSGEN    -DPLUGIN_GUI     -Isrc//default -c src//default/log.c -o src//default/log.o
cc -DGENCALL -DRUNTIME -g  -DPLUGIN_SCRIPTS   -Isrc//tmpasm  -DPLUGIN_PARSGEN    -DPLUGIN_GUI     -Isrc//default -c src//default/arg.c -o src//default/arg.o
cc -DGENCALL -DRUNTIME -g  -DPLUGIN_SCRIPTS   -Isrc//tmpasm  -DPLUGIN_PARSGEN    -DPLUGIN_GUI     -Isrc//default -c src//default/db.c -o src//default/db.o
cc -DGENCALL -DRUNTIME -g  -DPLUGIN_SCRIPTS   -Isrc//tmpasm  -DPLUGIN_PARSGEN    -DPLUGIN_GUI     -Isrc//default -c src//default/dep.c -o src//default/dep.o
cc -DGENCALL -DRUNTIME -g  -DPLUGIN_SCRIPTS   -Isrc//tmpasm  -DPLUGIN_PARSGEN    -DPLUGIN_GUI     -Isrc//default -c src//default/deps_default.c -o src//default/deps_default.o
cc -DGENCALL -DRUNTIME -g  -DPLUGIN_SCRIPTS   -Isrc//tmpasm  -DPLUGIN_PARSGEN    -DPLUGIN_GUI     -Isrc//default -c src//default/find_libs.c -o src//default/find_libs.o
cc -DGENCALL -DRUNTIME -g  -DPLUGIN_SCRIPTS   -Isrc//tmpasm  -DPLUGIN_PARSGEN    -DPLUGIN_GUI     -Isrc//default -c src//default/find_fscalls.c -o src//default/find_fscalls.o
cc -DGENCALL -DRUNTIME -g  -DPLUGIN_SCRIPTS   -Isrc//tmpasm  -DPLUGIN_PARSGEN    -DPLUGIN_GUI     -Isrc//default -c src//default/find_printf.c -o src//default/find_printf.o
cc -DGENCALL -DRUNTIME -g  -DPLUGIN_SCRIPTS   -Isrc//tmpasm  -DPLUGIN_PARSGEN    -DPLUGIN_GUI     -Isrc//default -c src//default/find_proc.c -o src//default/find_proc.o
cc -DGENCALL -DRUNTIME -g  -DPLUGIN_SCRIPTS   -Isrc//tmpasm  -DPLUGIN_PARSGEN    -DPLUGIN_GUI     -Isrc//default -c src//default/find_fstools.c -o src//default/find_fstools.o
cc -DGENCALL -DRUNTIME -g  -DPLUGIN_SCRIPTS   -Isrc//tmpasm  -DPLUGIN_PARSGEN    -DPLUGIN_GUI     -Isrc//default -c src//default/find_uname.c -o src//default/find_uname.o
cc -DGENCALL -DRUNTIME -g  -DPLUGIN_SCRIPTS   -Isrc//tmpasm  -DPLUGIN_PARSGEN    -DPLUGIN_GUI     -Isrc//default -c src//default/find_target.c -o src//default/find_target.o
cc -DGENCALL -DRUNTIME -g  -DPLUGIN_SCRIPTS   -Isrc//tmpasm  -DPLUGIN_PARSGEN    -DPLUGIN_GUI     -Isrc//default -c src//default/find_thread.c -o src//default/find_thread.o
cc -DGENCALL -DRUNTIME -g  -DPLUGIN_SCRIPTS   -Isrc//tmpasm  -DPLUGIN_PARSGEN    -DPLUGIN_GUI     -Isrc//default -c src//default/find_io.c -o src//default/find_io.o
cc -DGENCALL -DRUNTIME -g  -DPLUGIN_SCRIPTS   -Isrc//tmpasm  -DPLUGIN_PARSGEN    -DPLUGIN_GUI     -Isrc//default -c src//default/find_time.c -o src//default/find_time.o
cc -DGENCALL -DRUNTIME -g  -DPLUGIN_SCRIPTS   -Isrc//tmpasm  -DPLUGIN_PARSGEN    -DPLUGIN_GUI     -Isrc//default -c src//default/find_types.c -o src//default/find_types.o
cc -DGENCALL -DRUNTIME -g  -DPLUGIN_SCRIPTS   -Isrc//tmpasm  -DPLUGIN_PARSGEN    -DPLUGIN_GUI     -Isrc//default -c src//default/find_signal.c -o src//default/find_signal.o
cc -DGENCALL -DRUNTIME -g  -DPLUGIN_SCRIPTS   -Isrc//tmpasm  -DPLUGIN_PARSGEN    -DPLUGIN_GUI     -Isrc//default -c src//default/find_environ.c -o src//default/find_environ.o
cc -DGENCALL -DRUNTIME -g  -DPLUGIN_SCRIPTS   -Isrc//tmpasm  -DPLUGIN_PARSGEN    -DPLUGIN_GUI     -Isrc//default -c src//default/regex.c -o src//default/regex.o
cc -DGENCALL -DRUNTIME -g  -DPLUGIN_SCRIPTS   -Isrc//tmpasm  -DPLUGIN_PARSGEN    -DPLUGIN_GUI     -Isrc//default -c src//default/lib_filelist.c -o src//default/lib_filelist.o
cc -DGENCALL -DRUNTIME -g  -DPLUGIN_SCRIPTS   -Isrc//tmpasm  -DPLUGIN_PARSGEN    -DPLUGIN_GUI     -Isrc//default -c src//default/lib_srctree.c -o src//default/lib_srctree.o
cc -DGENCALL -DRUNTIME -g  -DPLUGIN_SCRIPTS   -Isrc//tmpasm  -DPLUGIN_PARSGEN    -DPLUGIN_GUI     -Isrc//default -c src//default/lib_pkg_config.c -o src//default/lib_pkg_config.o
cc -DGENCALL -DRUNTIME -g  -DPLUGIN_SCRIPTS   -Isrc//tmpasm  -DPLUGIN_PARSGEN    -DPLUGIN_GUI     -Isrc//default -c src//default/find_str.c -o src//default/find_str.o
cc -DGENCALL -DRUNTIME -g  -DPLUGIN_SCRIPTS   -Isrc//tmpasm  -DPLUGIN_PARSGEN    -DPLUGIN_GUI     -Isrc//default -c src//default/find_sys.c -o src//default/find_sys.o
cc -DGENCALL -DRUNTIME -g  -DPLUGIN_SCRIPTS   -Isrc//tmpasm  -DPLUGIN_PARSGEN    -DPLUGIN_GUI     -Isrc//default -c src//scripts/scripts.c -o src//scripts/scripts.o
cc -DGENCALL -DRUNTIME -g  -DPLUGIN_SCRIPTS   -Isrc//tmpasm  -DPLUGIN_PARSGEN    -DPLUGIN_GUI     -Isrc//default -c src//scripts/find_gpmi.c -o src//scripts/find_gpmi.o
cc -DGENCALL -DRUNTIME -g  -DPLUGIN_SCRIPTS   -Isrc//tmpasm  -DPLUGIN_PARSGEN    -DPLUGIN_GUI     -Isrc//default -c src//scripts/find_tcl.c -o src//scripts/find_tcl.o
cc -DGENCALL -DRUNTIME -g  -DPLUGIN_SCRIPTS   -Isrc//tmpasm  -DPLUGIN_PARSGEN    -DPLUGIN_GUI     -Isrc//default -c src//scripts/find_ruby.c -o src//scripts/find_ruby.o
cc -DGENCALL -DRUNTIME -g  -DPLUGIN_SCRIPTS   -Isrc//tmpasm  -DPLUGIN_PARSGEN    -DPLUGIN_GUI     -Isrc//default -c src//scripts/find_mruby.c -o src//scripts/find_mruby.o
cc -DGENCALL -DRUNTIME -g  -DPLUGIN_SCRIPTS   -Isrc//tmpasm  -DPLUGIN_PARSGEN    -DPLUGIN_GUI     -Isrc//default -c src//scripts/find_python.c -o src//scripts/find_python.o
cc -DGENCALL -DRUNTIME -g  -DPLUGIN_SCRIPTS   -Isrc//tmpasm  -DPLUGIN_PARSGEN    -DPLUGIN_GUI     -Isrc//default -c src//scripts/find_perl.c -o src//scripts/find_perl.o
cc -DGENCALL -DRUNTIME -g  -DPLUGIN_SCRIPTS   -Isrc//tmpasm  -DPLUGIN_PARSGEN    -DPLUGIN_GUI     -Isrc//default -c src//scripts/find_mawk.c -o src//scripts/find_mawk.o
cc -DGENCALL -DRUNTIME -g  -DPLUGIN_SCRIPTS   -Isrc//tmpasm  -DPLUGIN_PARSGEN    -DPLUGIN_GUI     -Isrc//default -c src//scripts/find_lua.c -o src//scripts/find_lua.o
cc -DGENCALL -DRUNTIME -g  -DPLUGIN_SCRIPTS   -Isrc//tmpasm  -DPLUGIN_PARSGEN    -DPLUGIN_GUI     -Isrc//default -c src//scripts/find_guile.c -o src//scripts/find_guile.o
cc -DGENCALL -DRUNTIME -g  -DPLUGIN_SCRIPTS   -Isrc//tmpasm  -DPLUGIN_PARSGEN    -DPLUGIN_GUI     -Isrc//default -c src//scripts/find_stutter.c -o src//scripts/find_stutter.o
cc -DGENCALL -DRUNTIME -g  -DPLUGIN_SCRIPTS   -Isrc//tmpasm  -DPLUGIN_PARSGEN    -DPLUGIN_GUI     -Isrc//default -c src//scripts/find_funlisp.c -o src//scripts/find_funlisp.o
cc -DGENCALL -DRUNTIME -g  -DPLUGIN_SCRIPTS   -Isrc//tmpasm  -DPLUGIN_PARSGEN    -DPLUGIN_GUI     -Isrc//default -c src//scripts/find_duktape.c -o src//scripts/find_duktape.o
[... cut ...]
make[1]: Leaving directory '/home/abhishek/xschem_git/src/utile'
abhishek@abhishek:~/xschem_git$ make install
cd src && make install
make[1]: Entering directory '/home/abhishek/xschem_git/src'
../scconfig/sccbox mkdir -p "/home/schippes/bin"
sccbox: No such file or directory
sccbox: failed to make directory with parents: /home/schippes/bin
make[1]: *** [Makefile:139: install] Error 255
make[1]: Leaving directory '/home/abhishek/xschem_git/src'
make: *** [Makefile:23: install] Error 2
abhishek@abhishek:~/xschem_git$

StefanSchippers, 2021-03-07:
You probably did run:
` ./configure ....--prefix=/home/schippes` .
schippes is my username on my system. You need to use your username, like: 
`./configure --prefix=/home/abhishek`

you can also run ./configure  with no prefix, this will install xschem in `/usr/local/bin` and xschem support files in `/usr/local/share`.
in this latter case you need to run make install with root privileges.
To solve the ussue do : 
```
make clean
./configure --prefix=/home/abhishek
make
make install
```

abhisupa, 2021-03-07:
Yeah, I followed exactly word by word :). I still seeing the same issue as before after following 
make clean
"./configure --prefix=/home/abhishek
make
make install" 

[image]

StefanSchippers, 2021-03-07:
The messages shown above are only warnings, xschem does build fine. I will eventually fix xschem to remove these warnings (i do not get any warning on my system, but this depend on the compiler)

abhisupa, 2021-03-07:
> The messages shown above are only warnings, xschem does build fine. I will eventually fix xschem to remove these warnings (i do not get any warning on my system, but this depend on the compiler)

Thanks Stefan, you are correct, I am able to run it now, Only difference from the video instruction to what I did is to run command 'sudo apt install xschem'. I thought that is not required once installed from git.

closed 2021-03-08 by abhisupa