StefanSchippers/xschem #11: Parameter handling in Verilog netlist generation

rbarzic, 2021-04-14:
Hello,

I'm having issues generating the correct output from the Verilog netlister when using parameter in a hierarchical context.
It seems that the netlister insists to add double-quotes (") around parameters when instanciating a module and refuse to add double-quotes around the default value in a parameter declaration
I've attached a small example 
- tb_xor.sch : the "top-vel" schematic
- tb_xor.sym - the associated symbol where  the default value for the parameter LIB is set (not sure that the best way to do it - maybe there is a better way to  do it - in the schematic ?)
- primitive_xor2.sym : an xor symbol . defining a LIB parameter, with a default value that should be LIB (without quote)
- tb_xor.v : the resulting verilog netlist
- tb_xor.expected.v : the expected netlist

Current result:
```verilog
module tb_xor (
  output wire out,
  input wire in0,
  input wire in1
);
parameter LIB = RTL4 ;
primitive_xor2
#(
.LIB ( "LIB" )
)
x1 ( 
 .Z( out ),
 .A( in0 ),
 .B( in1 )
);

endmodule
```
Expected result:

```verilog
module tb_xor (
  output wire out,
  input wire in0,
  input wire in1
);
parameter LIB = "RTL4" ;
primitive_xor2
#(
.LIB ( LIB )
)
x1 ( 
 .Z( out ),
 .A( in0 ),
 .B( in1 )
);

endmodule

```

[debug_xschem.gz](https://github.com/StefanSchippers/xschem/files/6309814/debug_xschem.gz)
(Note this is a tar.gz file but github only accept the extension .gz)

StefanSchippers, 2021-04-14:
Hi, Ronan, hope you are well.

To solve your problem:

- in primitive_xor2.sym, edit symbol attributes, delete the generic_type
  attribute altogether, this avoid the LIB parameter to be quoted.

- In tb_xor.sym, edit the template attribute as follows:
  template="name=x1 LIB=\\\"RTL4\\\""

Explanation:
Since 'template' value is double-quoted a backslash is used to
escape some characters (example: backslash itself, ")
so the value of 'template' after string parsing becomes:

name=x1 LIB=\"RTL4\"

This string is further processed to extract and separate 'name' and 'LIB' values,
The value of 'LIB' becomes: "RTL4".

if 'LIB' was defined as: LIB="RTL4" then 'LIB' would have been assigned
the value RTL4 (no double quotes)

This fixed the verilog netlist.

StefanSchippers, 2021-04-14:
A better approach is to delete the definition of the LIB parameter in tb_xor.sym
also delete `generic_type` and `extra` attributes in in primitive_xor2.sym
add the following in the xor2 instantiation ('q' on primitive_xor2.sym in the top schematic):
  name=x1 LIB=\"RTL4\"
When netlisting the LIB parameter will be correctly assigned to the xor2:
```
module tb_xor (
  output wire out,
  input wire in0,
  input wire in1
);

primitive_xor2
#(
.LIB ( "RTL4" )
)
x1 ( 
 .Z( out ),
 .A( in0 ),
 .B( in1 )
);

endmodule
```

closed 2021-04-20 by StefanSchippers