StefanSchippers/xschem #12: Feature Request: Symbol to Schematic mapping config

chrische-xx, 2021-04-16:
I found that when I work in xschem that I often like to reuse certain schematics.

 I have a complete testbench to evaluate an opamp. If I design another opamp I can duplicate the entire testbench of course and use it again. But now I have to maintain 2 identical testbenches where only the symbol of the opamp is exchanged. Now if I want to change something in the overall structure of the opamp testbench i need to to it twice.

It also happens when using hierarchical structures in schematic.

For example: I have a schematic hierarchy that uses a lot of switches. If I want to rerun the same simulation with an alternative switch schematic I need to replace every symbol with the new switch symbol everywhere in the hierarchy. That can potentially become cumbersome in a large schematic hierarchy with a lot of instances placed around.

So the feature request would be some sort of interface that allows to link a symbol in the hierarchy to an associated schematic. 

Btw. Its not really an issue, so feel free to close it again I was just not sure where to post this.

StefanSchippers, 2021-04-16:
Hi, 
What i do in these situations is to keep one testbench schematic, 
whenever i want to change the opamp i click the opamp, do a 'q' on it and change the reference to another opamp. If the two symbols have same pins you can go straight ahead with a new netlist and simulation.
There is another option to link a symbol to a different schematic. This is useful if you want to redirect many instances of a symbol to a different schematic implementation:
Descend into the symbol you want to redirect (click, press 'i') without selecting anything press 'q' and add this:
schematic=new_schematic.sch.
Picture below shows a symbol (`inv_1.sym`) to be redirected to `schematic inv_2.sch`:

[image]

chrische-xx, 2021-04-16:
Hi, Thanks for the quick and detailed reply. 

This is exactly the feature I was looking for.

closed 2021-04-16 by chrische-xx