StefanSchippers/xschem #13: Configure file error

ripudamank2, 2021-04-18:
Hi,
I was trying to configure the xschem but its giving below error
"make: *** No rule to make target 'src//scripts/find_welltype.c', needed by 'src//scripts/find_welltype.o'.  Stop.
./configure: 4: ./configure: not found"

I have install all the packages from below website.
http://repo.hu/projects/xschem/xschem_man/install_xschem.html

Can you please tell how to solve it.

[
[image]
](url)

StefanSchippers, 2021-04-18:
I am investigating. Will let you know asap...

StefanSchippers, 2021-04-18:
Some missing files have been checked in. Pull updates, should be ok now.
Thanks

closed 2021-04-18 by StefanSchippers