StefanSchippers/xschem #14: Request update to README_MacOS.md

stefanoaz, 2021-05-03:
In the README_MacOS.md file, the user is requested to modify the CFLAGS and LDFLAGS with particular options. I found that this caused my build to fail in Mac OS 10.15.7 (Catalina). Here is what I did for a successful build:

I installed the latest XQuartz from XQuartz.org.
I installed the latest tcl, tk and cairo from MacPorts. Porting from MacPorts made manual compilation of these libraries unnecessary.
I then followed Stefan's procedure to clone and configure.

However - I did *not* modify the makefile.conf. This is what the configure script produced in the makefile.conf, based on the libraries it found:

PREFIX=/Users/<PATH TO>/xschem-macos
CC=gcc
CFLAGS=-pipe -O2 -I/opt/local/include/cairo -I/opt/local/include/glib-2.0 -I/opt/local/lib/glib-2.0/include -I/opt/local/include -I/opt/local/include/pixman-1 -I/opt/local/include/ossp -I/opt/local/include/freetype2 -I/opt/local/include/libpng16 
LDFLAGS=-lm -L/opt/local/lib -lcairo -lX11 -lXrender -lxcb -lxcb-render -lX11-xcb -lXpm -ltcl8.6 -ltk8.6 

Leaving the makefile.conf untouched, I ran make and make install. This produced a successful build.

stefanoaz, 2021-05-03:
Additional comments for installing ngspice on MacOS, Running the autogen script with the adms flag, meant (of course) that I had to install ADMS. This was available as a port from MacPorts.

When I ran configure for ngspice, I needed to use these options:

../configure --enable-cider --enable-xspice --enable-openmp --enable-pss --enable-debug=no --prefix=/Users/azstefano/openTools/ngspice-macos --enable-openmp CFLAGS="-O2 -I/opt/X11/include/freetype2 -Wno-error=implicit-function-declaration" LDFLAGS="-lomp" --x-includes=/opt/X11/include --x-libraries=/opt/X11/lib

On the Mac, the clang and gcc compilers have -Werror set true, and this cannot be turned off in a global way. I found out that during the build, there were a few warnings about "implicit function declaration" which were converted to errors, stopping the build. By adding -Wno-error=implicit-function-declaration in the CFLAGS, the warnings remained warnings, and the build proceeded. 

(In my case, I got another error related to readline. To get rid of this error, I removed the option --enable-readline in the configure string.)

StefanSchippers, 2021-05-03:
Thank you Steve,
the README_MacOS was done for an xschem installation on MacOS 'Big Sur', so i think i will differentiate instructions for 'Catalina' and 'Big Sur'. Thank you very much for helping!.

stefanoaz, 2021-05-13:
My "Catalina build" did in fact require building tcl-tk from sources (MacPorts package manager). That and other special conditions are mentioned in the above paragraphs. My xschem is so far running normally, except that I cannot spawn a new schematic from an existing schematic window - I have posted the details as a separate bug report.

StefanSchippers, 2021-07-14:
README updated

closed 2021-07-14 by StefanSchippers