StefanSchippers/xschem #16: xschem installation issue

engrvip, 2021-05-17:
Hi,

After cloning from github and while trying to install xschem an error seen in attached screenshot is coming on my ubuntu machine. Please suggest how to resolve this in order to move ahead with xschem installation and other steps as shown in your youtube video.

Vipul

[image]

StefanSchippers, 2021-05-17:
you need to install the c compiler"
sudo apt-get install gcc
You also need to install all the required libraries as explained [here](http://repo.hu/projects/xschem/xschem_man/install_xschem.html).

engrvip, 2021-05-21:
Hi Stefan,
Thanks for your help on last issue. I am now stuck during ngspice installation. Have been following exact steps as given in youtube video. Issue shown in attached GIF file. Please suggest its resolution.

Regards
Vipul

[image]

benx45h, 2021-05-21:
You need adms, autoconf, libtool, and automake installed.
`sudo apt install adms autoconf libtool automake`

closed 2021-06-13 by StefanSchippers

Prabha1010, 2023-10-04:
Hi Stefan
I have installed xschem as per your instructions in your video 
after installation, I am unable to open xschem window. It appears as below
[image]

StefanSchippers, 2023-10-04:
This usually happen if no Xserver is running. Xschem uses the X11 graphic system. Try to run xclock or xterm (sudo apt install xterm) or any graphical application. You need the xorg packages and have a running X11 graphic server.
If xschem does not detect a running X11 server it switches to console mode. This is of course not very useful.

Prabha1010, 2023-10-24:
Tried with export command but it says
 X server connection failed, although DISPLAY shell variable is set.
   A possible reason is that the X server is not running or DISPLAY shell variable
   is incorrectly set.
   Starting Xschem in text only mode.

Prabha1010, 2023-10-24:
[image]

StefanSchippers, 2023-10-24:
The linux machine must be running the Xorg X11 server to allow xschem graphic mode. Is it running on your system?

Prabha1010, 2023-10-24:
How to check on this?

StefanSchippers, 2023-10-24:
dpkg -l xserver-xorg

Prabha1010, 2023-10-24:
[image]

StefanSchippers, 2023-10-24:
This explains the problem. No xserver is installed. Are you using an ubuntu install in a Windows WSL ? or do you have installed ubuntu as the main operating system on the computer?

Prabha1010, 2023-10-24:
Installed xserver-xorg. the error is still persisting

Prabha1010, 2023-10-24:
[image]

StefanSchippers, 2023-10-24:
I think you are on windows and using WSL, right?

Prabha1010, 2023-10-24:
> This explains the problem. No xserver is installed. Are you using an ubuntu install in a Windows WSL ? or do you have installed ubuntu as the main operating system on the computer?

I am running ubuntu on wsl

Prabha1010, 2023-10-24:
> I think you are on windows and using WSL, right?

yes

StefanSchippers, 2023-10-24:
do you have other ubuntu graphical applications running?

StefanSchippers, 2023-10-24:
I think you should take a look at this:
https://github.com/MicrosoftDocs/wsl/blob/main/WSL/tutorials/gui-apps.md

Prabha1010, 2023-10-24:
I tried with new drivers for my NVIDEA GeForce driver and still there is same issue. Is it something to do with ssh?

Prabha1010, 2023-10-24:
> do you have other ubuntu graphical applications running?

No

Prabha1010, 2023-10-24:
finally it is up and running 
[image]

thank you

siddhu679, 2023-10-27:
I have the same problem,
I am using wsl in windows
[image]

Prabha1010, 2023-10-27:
It's ./xschem

siddhu679, 2023-10-27:
[image]

.xshem file is not created

StefanSchippers, 2023-10-27:
the executable is in `~/bin/xschem`

siddhu679, 2023-10-27:
Thank you StefanSchippers,
I installed everything without any errors by following your video on YouTube, but when I want to open Xschem i get the following issues,

[image]

and when i click on netlist, simulations>edit netlist

[image]

StefanSchippers, 2023-10-27:
As the alert says there is no `gvim` edtor installed on the system. You can use another editor, see the xschemrc file `~/.xschem/xschemrc` and search for `editor`
You can use gedirt instead of gvim:
`set editor {gedit -w}`

StefanSchippers, 2023-10-27:
For the open_pdks issue as the warning says xschem did not find any in some common places, so you need to provide it in the PDK_ROOT environment variable:

export PDK_ROOT=/path/to/pdk
and then start xschem

siddhu679, 2023-10-28:
I resolved the above issues,
now when i click on simulate, i get this error
[image]

StefanSchippers, 2023-10-28:
Please follow all the steps [here](https://xschem.sourceforge.io/stefan/xschem_man/tutorial_xschem_sky130.html). Install `magic`, `xschem_sky130` and `open_pdks`, unless you installed a container with all the tools pre-installed. In this latter case contact who has distributed the container.

siddhu679, 2023-11-16:
I followed your video to install xschem, i have some doubts
Q1)what changes should i make in **xschemrc** file
file:///home/siddhuy/Pictures/Screenshots/Screenshot%20from%202023-11-16%2019-32-07.png

Q2)to provide PDK_ROOT environment variable, you mentioned this **export PDK_ROOT=/path/to/pdk** , can you tell me what is path to pdk.(i created same folder names that you mentioned in the video)

im getting this error
file:///home/siddhuy/Pictures/Screenshots/Screenshot%20from%202023-11-16%2019-35-45.png

Thank you

StefanSchippers, 2023-11-16:
you need to find the directory containing the pdk on your system: `...../pdk/sky130A/...`
When you have found it do this:
`export PDK_ROOT=...../pdk` 
Then you can install xschem normally.
Yu can set the PDK_ROOT in the xschemrc, just above line 9 in your image:

`set PDK_ROOT ..../pdk`

siddhu679, 2023-11-16:
Thank you for your time,
I installed everything that you mentioned in the video but i can't able to find any Pdk folder and sky130 folder.
 when i search pdk in my system i get the following folders
[image]

nndylsrknt, 2023-11-18:
I am getting the following error. I followed your video to install xschem.

[image]

nndylsrknt, 2023-11-18:
Actually I tried creating an example as you said in the video of open_pdks ( https://www.youtube.com/watch?v=KgBLByOkJxA&t=0s ),after installing xschem by using your first video https://www.youtube.com/watch?v=jXmmxO8WG8s 
But I am getting the following error when I am trying to open xschem.

[image]

StefanSchippers, 2023-11-18:
this means the ~/bin directory is not in your PATH search path.
just run xschem with `~/bin/xschem`

nndylsrknt, 2023-11-18:
Its working @StefanSchippers... If I follow the video( https://www.youtube.com/watch?v=KgBLByOkJxA&t=0s ) of open_pdks. Whether I need to do the patch for NgSpice as you did in your first video( https://www.youtube.com/watch?v=jXmmxO8WG8s ) for Spice installation...because I couldn't find any xschem_library  folder in .xschem folder ?

StefanSchippers, 2023-11-18:
No the patch is no more needed. Just follow the second video. Open_pdks installation now handles everything for the pdk and spice simulation models. FOr further reference use[ this manual page](https://xschem.sourceforge.io/stefan/xschem_man/tutorial_xschem_sky130.html).

nndylsrknt, 2023-11-18:
But before proceeding with the spice installation whether I need to do install skywater130 under xschem_library ? Because in the second video we have no where linked skywater and xschem ?

StefanSchippers, 2023-11-18:
If you succesfully install open_pdks it will install also all the xschem components and schematics for using sky130.
Then you should crete a test directory, copy the template xschemrc into it and run xschem from thai directory.
Please follow very carefully all the steps explained in [this manual page](https://xschem.sourceforge.io/stefan/xschem_man/tutorial_xschem_sky130.html), the section containing the text below:

...
...
 After completing the above steps you can do a test run of xschem and use the Sky130 devices. You need to create a new empty drectory, copy the provided xschemrc (<prefix>/share/pdk/sky130B/libs.tech/xschem/xschemrc) into it and run xschem:

```
mkdir test_xschem_sky130
cd test_xschem_sky130
cp /usr/local/share/pdk/sky130B/libs.tech/xschem/xschemrc .
xschem
```
...
...

nndylsrknt, 2023-11-18:
Hi @StefanSchippers... I am trying to install NgSpice now, by following the steps provided in the first video, but there is no Simulations folder inside .Xschem folder. What could be the reason/fix for this?

[image]

nndylsrknt, 2023-11-18:
Hi @StefanSchippers...I am getting the following error when I am trying to simulate a test circuit.

[image]

StefanSchippers, 2023-11-18:
This happens because /home/user/bin is not in your path variable.
in Xschem, go to Simulation->Configure simulators and tools and replace `ngspice` with `/home/your_user/bin/ngspice`

[image]

nndylsrknt, 2023-11-18:
Hi @StefanSchippers...everything is working good now...but the simulation results are closing themselves very immediately, they are not coming as pop-ups. Anything to do with configure simulators and tools? 

[image]

StefanSchippers, 2023-11-18:
The simulator window is closed at the end of the simulation because there is a `quit 0` command in the code block. Select the instance with simulator commands, press 'q' and remove the `quit 0` line.

nndylsrknt, 2023-11-18:
Hi @StefanSchippers...can you please brief more? what is the exact file you are mentioning about?

nndylsrknt, 2023-11-18:
@StefanSchippers...got it! Thanks a lot!!

uzairmirza, 2023-12-03:
> finally it is up and running <img alt="image" width="725" src="https://private-user-images.githubusercontent.com/61615252/277695519-e43029a6-6cc2-40ac-b364-8b93655aac78.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDE1ODgyNzgsIm5iZiI6MTcwMTU4Nzk3OCwicGF0aCI6Ii82MTYxNTI1Mi8yNzc2OTU1MTktZTQzMDI5YTYtNmNjMi00MGFjLWIzNjQtOGI5MzY1NWFhYzc4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMDMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjAzVDA3MTkzOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWViODUzMTc2ZTk1YWQ5MDU3YjdmNWE5ZDhmMjhhMDc1MGMzYmJkM2EyOGFlYmFmOWU1MGM1MjQ0ZmM3ZDQ3NDcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.qNF-StTpI8OOPJ08NkYtbi3iRwE06IQwBnIT2RaN7zc">
> 
> thank you

Hello Prabha! I am facing the same issue. How you have resolved this issue?

ucifer6, 2023-12-04:
i am alway find this type of error. when i am using gvim editor and i am also not get to shown the netlist window in option bar. please some one help me i am a new user.

[image]
[image]
 
can not execute gvim-f {-c :set filetype=spice} /tmp/.xschem_edit_file.23952.v':
ensusre it is available on the system.
error: could't execute "gvim":no such file or directory solve this issue in my xschem

ucifer6, 2023-12-04:
[image]

ucifer6, 2023-12-04:
i am alway find this type of error. when i am using gvim editor and i am also not get to shown the netlist window in option bar. please some one help me i am a new user.

can not execute gvim-f {-c :set filetype=spice} /tmp/.xschem_edit_file.23952.v':
ensusre it is available on the system.
error: could't execute "gvim":no such file or directory solve this issue in my xschem

Prabha1010, 2023-12-05:
Hello all
Please find out the required graphics driver available for your PC or laptop
Install and run
In my case Nvidia driver solved the issue

siddhu679, 2023-12-08:
Hello @StefanSchippers, while simulating the **tb_reram**(which is already present in the analog primitives) i get the following error like "sky130_fd_pr_reram_reram_model unknown device type -error". Can you please help me with this.
i attach the error screenshot
[image]

StefanSchippers, 2023-12-08:
If ngspice does not recognize the n device (verilogA model) this means it is an older build that did not support osdi compiled verilog-A devices. You need to clone the ngspice sources and configure with option  --enable-osdi.
Second, you must install openvaf, compile the reram verilog-A code and place the resulting compiled file into 
$SKYWATER_MODELS/sky130_fd_pr_reram__reram_module.osdi

siddhu679, 2023-12-09:
@StefanSchippers, Can you please elaborate further

siddhu679, 2023-12-12:
@StefanSchippers can you please tell me how to install openvaf and compile the reram verilog-A code, I need to work on reream device.

StefanSchippers, 2023-12-13:
- download this file: https://datashare.tu-dresden.de/s/deELsiBGyitSS3o
- extract the openvaf executable: t`ar zxvf openvaf_23_5_0_linux_amd64.tar.gz`
- make the file executable: `chmod a+x openvaf`
- get the reram verilogA module:` sky130_fd_pr_reram__reram_module.va`
- compile the module:  ` ./openvaf sky130_fd_pr_reram__reram_module.va`
- put the resulting `sky130_fd_pr_reram__reram_module.osdi` file into your pdk:
- `cp sky130_fd_pr_reram__reram_module.osdi .../pdk/sky130B/libs.tech/ngspice/sky130_fd_pr_reram__reram_module.osdi`

Usually you should use the sky130B process variant, however to make things simple copy the osdi file also in sky130A:
- `cp sky130_fd_pr_reram__reram_module.osdi .../pdk/sky130A/libs.tech/ngspice/sky130_fd_pr_reram__reram_module.osdi`

siddhu679, 2023-12-13:
Thank you, @StefanSchippers. Could you guide me on acquiring the RERAM Verilog-A module? I'm unable to locate it in my system—should I download it from the browser?

nndylsrknt, 2023-12-18:
Hi @StefanSchippers... I am getting the following error when trying to simulate the tb_RERAM. What could be the reason/fix for this?

[image]

[19 more replies]