StefanSchippers/xschem #18: incorrect rendering of parameters in verilog

bjmuld, 2021-06-21:
symbol properties' "template" field not rendering correctly in generated Verilog.
See skywater: `pfet_g5v0d10v5` for example.
Template is defined:

```
template="name=M1
L=0.5
W=1
nf=1
mult=1
ad=\"'int((nf+1)/2) * W/nf * 0.29'\" 
pd=\"'2*int((nf+1)/2) * (W/nf + 0.29)'\"
as=\"'int((nf+2)/2) * W/nf * 0.29'\" 
ps=\"'2*int((nf+2)/2) * (W/nf + 0.29)'\"
nrd=\"'0.29 / W'\" nrs=\"'0.29 / W'\"
sa=0 sb=0 sd=0
model=pfet_g5v0d10v5
spiceprefix=X
"
```

And Verilog produced is invalid (note the quoting in the ad, pd, as, ps parameters):
```
pfet_g5v0d10v5
#(
.L ( 0.5 ) ,
.W ( 1 ) ,
.nf ( 1 ) ,
.mult ( 1 ) ,
.ad ( "'int((nf+1)/2) ) ,
.pd ( "'2*int((nf+1)/2) ) ,
.as ( "'int((nf+2)/2) ) ,
.ps ( "'2*int((nf+2)/2) ) ,
.nrd ( "'0.29 ) ,
.nrs ( "'0.29 ) ,
.sa ( 0 ) ,
.sb ( 0 ) ,
.sd ( 0 ) ,
.model ( pfet_g5v0d10v5 ) ,
.spiceprefix ( X )
)
M3 ( 
 .D( Row4 ),
 .G( net1 ),
 .S( Col1 ),
 .B( Vinj )
);
```

StefanSchippers, 2021-06-21:
The definitions for mos transistors are for a spice simulator. To do a verilog simulation you can easily add to the transistor symbol a netlisting rule for verilog:

in the mos symbol edit the global atributes and insert the following:

verilog_format="pmos #(@risedel , @falldel , @offdel ) @name ( @@d , @@s , @@g );"

see following [video](https://xschem.sourceforge.io/stefan/xschem_man/video_tutorials/multiple_schematics_bound_to_symbol.mp4) about the subject:
See also attached image for a pmos definition that allows verilog simulation:
[image]

bjmuld, 2021-06-21:
Is this desired behavior?  Since the `verilog_format` field exists, why should the `format` field's values be used in verilog generation?

I anticipate wanting to swich back and forth between netlisting a circuit to spice for spice simulation, and to veriog for synthesis.  Seems like we'd want spice parameters to appear in spice netlists only and verilog parameters appearing in verilog only. No?

StefanSchippers, 2021-06-21:
Normally symbols containing transistors in the schematic should either have:
- a behavioral code that allows the verilog simulator to simulate the component without descending to the schematic.
- simulating mos transistors using verilog builtin 'switch' primitives.

If you are netlisting a symbol that has a schematic with transistors, what the netlister does is to create an instance call of the transistor (verilog does not use the format attribute that defines the netlisting rule for spice only):
symbol name - parameter list - instance name - connection list.
The parameter list is taken from the list of the parameters in the instance. If you click on a transistor, press 'q' and delete the ad=..., as=..., ps=..., pd=... attributes or give plain numbers to these attributes the verilog netlist will be syntactically correct.
However a verilog simulator does not know what to do with all these numbers.

If you open the example schematic provided with xschem: 
examples/greycnt.sch, this design can be simulated without changes in spice, vhdl and verilog. You can take this as an example.
[image]

StefanSchippers, 2021-07-14:
Not a bug

closed 2021-07-14 by StefanSchippers