StefanSchippers/xschem #19: simulation fail, showing a lots of warnings and an error

pauloohaha, 2021-07-26:
Hi,
I have followed your detailed video on youtube to install xschem on my ubuntu system in a virtual machine run on windows.
I have finished all the installations. However, when I tried to run the simulation, I met some problems.

The first problem is already solved, but I think maybe it is good to let you know:
When I ran the simulation, it showed this:
[image]

I consult with one of my friends, who have also met this problem and sent you an email, and solved the problem by modifing the path in nfet_20v0_MODEL and TT MODELS:
[image]

Then I met the second problem, which is the main problem that I wish to get help from you:
Afther the paths are modified, I ran the simulation again. But this time it stuck at here for about 1min:
[image]

Then it shows a lots of warnings and stuck again:
[image]

After few mins, it shows an error:
[image]

I have looked up in internet, but it seems that there is no existing solutions (if I didn't missed it

Anyway, thanks for your contribution and help!!!

StefanSchippers, 2021-07-26:
you must replace the `.include /path/to/sky130.lib.spice` with:
`.lib /path/to/sky130.lib.spice tt`

Stefan

pauloohaha, 2021-07-27:
Let me show the content of my nfet_20v0_MODEL and TT_MODELS:

nfet_20v0_MODEL:
[image]

TT_MODELS:
[image]

 I have also creadted the .spiceinit file.

The situation is now:
If I start simulation, it pop up a lots of messages (a small part of them):
[image]
and then:
[image]

If I type y and then enter:
[image]

It seems that something is still wrong.

StefanSchippers, 2021-07-27:
If you use directly the `sky130_fd_pr` tree you probably need to apply this [patch](https://github.com/StefanSchippers/xschem_sky130/blob/main/sky130_fd_pr.patch) file.
Download the [file](https://raw.githubusercontent.com/StefanSchippers/xschem_sky130/main/sky130_fd_pr.patch) from github, call it `sky130_fd_pr.patch`, go into the `.../sky130_fd_pr/latest/` directory and execute:

`patch -p2 < sky130_fd_pr.patch`

pauloohaha, 2021-07-27:
I copied the `sky130_fd_pr.patch` file from the git clone during the installation and put it in the `.../sky130_fd_pr/latest/` directory.

Then I ran `patch -p2 < sky130_fd_pr.patch` and getting this:
[image]

StefanSchippers, 2021-07-27:
Only a single file failed patching, this is probably because there is a newer version of the skywater pdk,. No problem, 
Run simulation and see if errors are now gone.

pauloohaha, 2021-07-27:
Good news, some graphs pop up:
[image]

But are them kind of strange? I think there is still something wrong.

The message from the simulation:
[image]

There are a lot of 'ignored' messages:
[image]

Message at last:
[image]

StefanSchippers, 2021-07-27:
Your ngspice version is too old, 
clone ngspice from [git](https://git.code.sf.net/p/ngspice/ngspice), go into the ngspice/ directory, switch to pre-master branch (`git checkout pre-master`) and follow the instructions in the INSTALL file.

you need to run `./autogen.sh`
then create a `release/` directory under the `ngspice/` tree, cd to `release/`, and run:
`../configure --with-x --enable-xspice --disable-debug --enable-cider --with-readline=yes`
followed by '`make`' and '`make install`'

pauloohaha, 2021-07-27:
Everything went well!
[image]

Huge thank for your generous and timely help!!!!

(b.t.w. I guess from your name that you maybe a German and Vielen Dank!!!!

closed 2021-07-27 by pauloohaha