StefanSchippers/xschem #20: Simulation problem in xschem

ripudamank2, 2021-08-01:
The simulation is not able to map the directory properly.  I tried different examples every time same problem. 
I checked your video and setup like that but still the same problem.
Can you please help to solve this?

.include /home/xxx/Documents/skywater-130nm/libraries/sky130_fd_pr_ngspice/latest/../../libs.ref/sky130_fd_pr/spice/sky130_fd_pr__pfet_20v0__tt_discrete.corner.spice

 .lib /home/xxxx/Documents/skywater-130nm/libraries/sky130_fd_pr_ngspice/latest/sky130.lib.spice tt

StefanSchippers, 2021-08-03:
The skywater pdk is evolving, and now xschem paths follow the open-pdks conventions.
What is recently changed is that open-pdks now installs xschem libraries and builds and patches the spice models in a different directory structure. If you installed skywater-pdks as explained in the video, just locate the two files on your system (use the find command) and change the .lib and .include lines in the attributes of the TT_MODELS and pfet_20v0_MODEL components:

like this:

.lib /full/path/of/sky130.lib.spice tt

and 

.include /full/path/of/sky130_fd_pr__pfet_20v0__tt_discrete.corner.spice

with the correct paths ngspic will be able to complete the simulation/

ripudamank2, 2021-08-24:
I tried searching these 2 files in  TT_MODELS and pfet_20v0_MODEL . 
In my system using find command but not able to find it . 
Can you please help me out . 
thanks

ripudamank2, 2021-08-24:
Thanks a lot . Its working fine

closed 2021-08-24 by ripudamank2