StefanSchippers/xschem #22: [UI/UX Improvement] Allow going deeper into the circuit.

armleo, 2021-09-14:
UI/UX improvement: When working with top schematics that contains sub-schematics it's hard to navigate between multiple sub-schematics, every time you have to use `File -> Open` to access the schematic of the component. Instead, allow to double click the component to go deeper into the hierarchy and add buttons to leave "deeper" hierarchy up to top level.

StefanSchippers, 2021-09-14:
to descend into a component click the component (make it selected), then press the 'e' key (Edit -> Push Schematic). To return back to parent press 'ctrl-e'

You can press 'Alt-e' instead of 'e' if you want to open the sub-schematic in a new window.

armleo, 2021-09-15:
This is great, thanks, I thought there was no way to do it!

closed 2021-09-15 by armleo