StefanSchippers/xschem #23: Alter the netlist between GUI or script

chrische-xx, 2021-09-19:
I'm trying to netlist certain blocks only when hitting the netlist button in the GUI and ignore them when running xschem from the commandline with the --netlist option.

Any advise on how to achieve this?

StefanSchippers, 2021-09-19:
GUI button and command line netlist invocation do exactly the same thing, however i have made a new commit where *any* attribute can be wrapped into a `tcleval(....)` construct and attribute will be passed to tcl for evaluation. this allow any kind of conditional execution. For your use case:

xschem uses a `spice_ignore` attribute to decide if a symbol must be ignored completely when netlisting. If `spice_ignore=true` the symbol will not be descended and not netlisted at all.
So if you want some symbols to be excluded, descend into the symbol (click to select and press 'i' key), then press 'q' to edit the symbol global attributes and add the following line:
`spice_ignore="tcleval($::cmdline_ignore)"`
if the tcl global variable `cmdline_ignore` is not set this will evaluate to the empty string and xschem will netlist the symbol.
however if in the xschem window you write: 
`set cmdline_ignore true`
any following netlist operation will exclude all symbols that have the above attribute.

now if you enable xschem on command line in the following way: 
`xschem   -n -s -q --tcl 'set cmdline_ignore false'  your_schematic.sch`
it will skip all symbols that have the `spice_ignore="tcleval($::cmdline_ignore)"` attribute.
If you invoke xschem from GUI and use the netlist button all symbols will be netlisted unless you set the tcl variable.
You can use any unused tcl variable, `cmdline_ignore` is just my example.

chrische-xx, 2021-09-20:
Excellent! This is exactly what i needed.

In case someone else will look into this issue: This can also be applied selectively to individual symbols in a schematic. 

I have for example a usecase where I have two "code.sym" symbols in my schematic. One of the two defines the ngspice controls, while the other includes the simulation corner. 
When running corner simulation from an external script i will automatically update the netlist with the right corner setup, so i do not want to netlist the code block containing the corner setup. 

Thanks again for the fast reply!

StefanSchippers, 2021-09-20:
Thanks for opening this issue. I remember you asked some time ago how to map a schematic to a symbo in a similar wayl, at the time i made the change only for the 'schematic' attribute, now the 'tcleval(...)' is universal, can be applied to any attribute. This makes the feature very appealing, allowing conditional execution based on tcl variables (or commands ) or shell variables (via tcl $env(VAR) ).
Some testing needs to be done, i am running regression tests to ensure nothing is broken due to the change. If everything is ok i will eventually close the issue.

StefanSchippers, 2021-10-05:
no issues in tests. closed

closed 2021-10-05 by StefanSchippers