StefanSchippers/xschem #24: Build on MacOS Big Sur not working

hpretl, 2021-09-23:
I am following the MacOS build instructions in `README_MacOS.md` for Big Sur, with one change: I build Tcl and Tk into /usr/local/opt2/tcl-tk, because I have already a Tcl-Tk installation in /usr/local/opt done via HomeBrew.

Compile of Tcl, Tk and xschem runs w/o error, but when I start `xschem` I get the following error:

```
dyld: Library not loaded: /usr/local/opt2/tcl-tk/lib:/opt/X11/lib/libtk8.6.dylib
  Referenced from: /Users/.../xschem-macos/bin/./xschem
  Reason: image not found
zsh: abort      ./xschem
```

If I try to use the Tcl/Tk from HomeBrew, I get the following error:

```
EMERGENCY SAVE DIR: /tmp/xschem_emergencysave__foeRfFlgd9

FATAL: signal 11
while editing: 
```

It would be ideal, if the compile of `xschem` would work with Tcl-Tk from HomeBrew. Even better would an install via HomeBrew also for xschem :-)

StefanSchippers, 2021-09-23:
After building tcl-tk did you run xschem configure and before running 'make' change `Makefile.conf `changing the CFLAGS and LDFLAGS?
CFLAGS=-I/opt/X11/include -I/opt/X11/include/cairo \
**-I/usr/local/opt2/tcl-tk/include** -O2
LDFLAGS=-L/opt/X11/lib **-L/usr/local/opt2/tcl-tk/lib** -lm -lcairo \
-lX11 -lXrender -lxcb -lxcb-render -lX11-xcb -lXpm -ltcl8.6 -ltk8.6

Xschem can not run with native macOS tcl-tk libraries, as native graphics on macOS is Quartz and not X11, in order to run xschem (which is an X11 application) you need to use the xquartz layer and tcl-tk built and linked with xquartz.
Unfortunately i do not have a macOS system so the info i can give may not be complete, however there are Big Sur users who successfully built xschem.

StefanSchippers, 2021-09-23:
doesn't the 'image not found' relate to some missing files in the /usr/local/opt2/tcl-tk directories?

StefanSchippers, 2021-09-23:
Some similar usses apprently solved by running some 'brew' commands, may be it's worth giving a try here:
https://try2explore.com/questions/10039989

StefanSchippers, 2021-09-23:
Looking at the error message there is some syntax error somewhere, there is an incorrect path:
dyld: Library not loaded: **/usr/local/opt2/tcl-tk/lib:/opt/X11/lib/libtk8.6.dylib**

hpretl, 2021-09-23:
> Looking at the error message there is some syntax error somewhere, there is an incorrect path:
> dyld: Library not loaded: /usr/local/opt2/tcl-tk/lib:/opt/X11/lib/libtk8.6.dylib
> it should probably be: /usr/local/opt2/tcl-tk/lib/libtk8.6.dylib

Yes, this was my guess that this is the issue, however, I have no idea how this could have happened. Yes, I did as in the README:

1. build Tcl and Tk locally
2. run ./configure of xschem
3. fix the Makefile.conf according to your instructions

I attach the Makefile.conf here. Somehow it looks like the linker merges both `-L`and applies it?

[Makefile.conf.gz](https://github.com/StefanSchippers/xschem/files/7218844/Makefile.conf.gz)

StefanSchippers, 2021-09-23:
Try to run otool on the xschem executable, like (this was done by a macOS user):
➜  xschem git:(master) ✗ otool -L ./src/xschem
./src/xschem:
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1292.0.0)
	/opt/X11/lib/libcairo.2.dylib (compatibility version 11403.0.0, current version 11403.2.0)
	/opt/X11/lib/libX11.6.dylib (compatibility version 10.0.0, current version 10.0.0)
	/opt/X11/lib/libXrender.1.dylib (compatibility version 5.0.0, current version 5.0.0)
	/opt/X11/lib/libxcb.1.dylib (compatibility version 3.0.0, current version 3.0.0)
	/opt/X11/lib/libxcb-render.0.dylib (compatibility version 1.0.0, current version 1.0.0)
	/opt/X11/lib/libX11-xcb.1.dylib (compatibility version 2.0.0, current version 2.0.0)
	/opt/X11/lib/libXpm.4.dylib (compatibility version 16.0.0, current version 16.0.0)
	/usr/local/opt/tcl-tk/lib/libtcl8.6.dylib (compatibility version 8.6.0, current version 8.6.10)
	/usr/local/opt/tcl-tk/lib/libtk8.6.dylib (compatibility version 8.6.0, current version 8.6.10)

StefanSchippers, 2021-09-23:
try also to do a 'make distclean' in xschem source directory and run configure-edit Makefile.conf-make-make install again...

hpretl, 2021-09-24:
I did a bit more debugging, focusing on the HomeBrew tcl-tk installation, as this is proven to work with X11(XQuartz).

When I use `otool` to inspect the compiled binary I get this, so all good:

```
% otool -L xschem
xschem:
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1292.100.5)
	/usr/local/opt/cairo/lib/libcairo.2.dylib (compatibility version 11603.0.0, current version 11603.0.0)
	/usr/local/opt/libx11/lib/libX11.6.dylib (compatibility version 11.0.0, current version 11.0.0)
	/usr/local/opt/libxrender/lib/libXrender.1.dylib (compatibility version 5.0.0, current version 5.0.0)
	/usr/local/opt/libxcb/lib/libxcb.1.dylib (compatibility version 3.0.0, current version 3.0.0)
	/usr/local/opt/libxcb/lib/libxcb-render.0.dylib (compatibility version 1.0.0, current version 1.0.0)
	/usr/local/opt/libx11/lib/libX11-xcb.1.dylib (compatibility version 2.0.0, current version 2.0.0)
	/usr/local/opt/libxpm/lib/libXpm.4.dylib (compatibility version 16.0.0, current version 16.0.0)
	**/usr/local/opt/tcl-tk/lib/libtcl8.6.dylib (compatibility version 8.6.0, current version 8.6.11)**
	**/usr/local/opt/tcl-tk/lib/libtk8.6.dylib (compatibility version 8.6.0, current version 8.6.11)**
```

When I check the path, it looks also good, the tcl-tk from HomeBrew in /opt/local comes first:

```
% echo $path
/usr/local/opt/tcl-tk/bin /usr/local/opt/python@3.9/bin /usr/local/sbin /usr/local/bin /usr/local/texlive/2021basic/bin/universal-darwin /usr/local/bin /usr/bin /bin /usr/sbin /sbin /Library/TeX/texbin /opt/X11/bin /Library/Apple/usr/bin /Library/Frameworks/Mono.framework/Versions/Current/Commands
```

Here is the debug info leading up to the crash, maybe this helps:

```
% ./xschem -d 1
Tcl_AppInit(): sourcing /Users/___/src/xschem/src/xschemrc
Tcl_AppInit(): XSCHEM_SHAREDIR=/Users/___/src/xschem/src  XSCHEM_LIBRARY_PATH=/Users/___/src/xschem/xschem_library/devices
Tcl_AppInit(): done step a of xinit()
Tcl_AppInit(): done step a1 of xinit()
Tcl_AppInit(): sourcing /Users/___/src/xschem/src/xschem.tcl
Tcl_AppInit(): done executing xschem.tcl
Tcl_AppInit(): resolved xschem_executable=/usr/local/bin/./xschem
create_tmpdir(): created dir: /tmp/xschem_undo_8m1K5hNbS1
undo_dirname=/tmp/xschem_undo_8m1K5hNbS1
remove_symbols(): done
load_schematic(): opening file for loading:/Users/___/src/xschem/src/systemlib/font.sch, filename=/Users/harald/src/xschem/src/systemlib/font.sch
load_schematic(): sch[currsch]=/Users/___/src/xschem/src/systemlib/font.sch
del_inst_table(): cleared object hash table
clear drawing(): deleted data structures, now deleting hash
inst_free_hash(): removing hash table
load_schematic(): reading file: /Users/___/src/xschem/src/systemlib/font.sch
read_xschem_file(): file_version=1.2
set_modify(): 0
load_schematic(): /Users/___/src/xschem/src/systemlib/font.sch, returning
del_inst_table(): cleared object hash table
clear drawing(): deleted data structures, now deleting hash
inst_free_hash(): removing hash table
Tcl_AppInit(): drawing window ID=0x7f9808a1d810
Tcl_AppInit(): top window ID=0x7f980a046d10
Tcl_AppInit(): done tkinit()
create_tmpdir(): created dir: /tmp/xschem_emergencysave__BZo7Ullb6Y
EMERGENCY SAVE DIR: /tmp/xschem_emergencysave__BZo7Ullb6Y

FATAL: signal 11
while editing: 
```

hpretl, 2021-09-24:
> try also to do a 'make distclean' in xschem source directory and run configure-edit Makefile.conf-make-make install again...

Tried it, no change, still crash.

StefanSchippers, 2021-09-24:
Try this:
After configure edit also config.h and set HAS_XCB and HAS_XRENDER to zero:
```
...
/* Define this var if Xlib's xcb is available */
#define HAS_XCB 0

/* Define this var if Xlib's xrender is available */
#define HAS_XRENDER 0
...
```
xschem runs without degradation even without these extensions.

StefanSchippers, 2021-09-24:
If the tcl-tk install in /usr/local/opt/tcl-tk is for x11 then there should be also a bin directory (/usr/local/opt/tcl-tk/bin ? ) where the 'wish' or 'wish8.6' programs are available.
try to execute that wish program and give the following command:
**winfo server .**
if the tcl-tk interpreter is built for x11 you should get: 
X11R0 The X.Org <http://X.Org> Foundation 12008000
or something similar.

hpretl, 2021-09-25:
It works now, I found the bug: Your instructions in `README_MacOS.md` are correct, and I followed them (I just used the path `/usr/local/opt2/tcl-tk` so that this new `tcl-tk` does not interfere with the version from HomeBrew).

The issue is caused by a bug in the `tk` build script, the wrong path is in `wish8.6`. In the `Makefile` of `tk8.6.10` the following needs to be changed:

```
LIB_RUNTIME_DIR          = $(libdir):/opt/X11/lib
```

to 

```
LIB_RUNTIME_DIR         = $(libdir)
```

What I still don't understand: When I use the tcl-tk from HomeBrew, there is still a crash with signal 11?

StefanSchippers, 2021-09-26:
i know xschem will crash with signal 11 if run with a tcl-tk version for another graphic system. But if you are sure the tcl-tk libraries you are using runtime are built for X11 Xquartz this is strange...

hpretl, 2021-09-26:
Yes, that is the reason. Here is a comparison of the HomeBrew `wish8.6` and the custom-made one. The HomeBrew one uses Cocoa, the custom-made X11.

```
% otool -L /usr/local/opt/tcl-tk/bin/wish8.6
/usr/local/opt/tcl-tk/bin/wish8.6:
	/usr/local/Cellar/tcl-tk/8.6.11_1/lib/libtk8.6.dylib (compatibility version 8.6.0, current version 8.6.11)
	/usr/local/Cellar/tcl-tk/8.6.11_1/lib/libtcl8.6.dylib (compatibility version 8.6.0, current version 8.6.11)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1292.100.5)
	/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation (compatibility version 150.0.0, current version 1775.118.101)
	/System/Library/Frameworks/Cocoa.framework/Versions/A/Cocoa (compatibility version 1.0.0, current version 23.0.0)
	/System/Library/Frameworks/Carbon.framework/Versions/A/Carbon (compatibility version 2.0.0, current version 164.0.0)
	/System/Library/Frameworks/IOKit.framework/Versions/A/IOKit (compatibility version 1.0.0, current version 275.0.0)
	/usr/lib/libz.1.dylib (compatibility version 1.0.0, current version 1.2.11)
```

```
otool -L /usr/local/opt2/tcl-tk/bin/wish8.6
/usr/local/opt2/tcl-tk/bin/wish8.6:
	/usr/local/opt2/tcl-tk/lib/libtk8.6.dylib (compatibility version 8.6.0, current version 8.6.10)
	/usr/local/opt2/tcl-tk/lib/libtcl8.6.dylib (compatibility version 8.6.0, current version 8.6.10)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1292.100.5)
	/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation (compatibility version 150.0.0, current version 1775.118.101)
	/opt/X11/lib/libX11.6.dylib (compatibility version 11.0.0, current version 11.0.0)
	/opt/X11/lib/libXss.1.dylib (compatibility version 2.0.0, current version 2.0.0, weak)
	/opt/X11/lib/libXext.6.dylib (compatibility version 11.0.0, current version 11.0.0)
	/usr/lib/libz.1.dylib (compatibility version 1.0.0, current version 1.2.11)
```

StefanSchippers, 2021-09-26:
This is indeed what i knew. You must build and compile xschem with the /opt2 tcl-tk version, and also ensure that when running xschem the dynamic linker will use the /opt2/ libraries.

hpretl, 2021-09-26:
BTW, do you have any plans to provide a HomeBrew recipe? This would be ideal for MacOS users, as `ngspice` and `gtkwave` are already available.

StefanSchippers, 2021-09-26:
I know for sure some Big sur MacOS users were able to build and run xschem (see image) using the x11 version of tcl-tk + xquartz
For a native "homebrew" i need a macOS developer. This has happened for windows, now there is a [native xschem for windows](http://repo.hu/projects/xschem/releases-bin/).

[image]

hpretl, 2021-09-26:
It works now on my machine (see earlier post "It works now, I found the bug...") after fixing the bug in the `tk` compilation. I put creating a HomeBrew recipe for `xschem` on my todo-list, but I can't promise a date.

closed 2021-09-26 by hpretl

StefanSchippers, 2021-09-26:
Oh, thank you, i thought the "it works now" was only related to compiling, but still not running. 
If there are steps you did that are worth adding in the README for MacOS please le me know, i will make the additions, to avoid the xschem build frustration in other MacOS users. Thank you !

StefanSchippers, 2021-09-26:
Will add a note about the change in Tk Makefile as you suggested...

The issue is caused by a bug in the tk build script, the wrong path is in wish8.6. In the Makefile of tk8.6.10 the following needs to be changed:

LIB_RUNTIME_DIR          = $(libdir):/opt/X11/lib

to

LIB_RUNTIME_DIR         = $(libdir)

StefanSchippers, 2021-09-26:
by the way i have a tck8.4 installation. This is a 20 years old tcl-tk. I use this to ensure xschem still runs with these old libraries, and the Makefile for tk shows indeed:

# Path to use at runtime to refer to LIB_INSTALL_DIR:
LIB_RUNTIME_DIR         = $(libdir)

hpretl, 2021-09-26:
Maybe also add a note to use `/usr/local/opt2` instead of `/usr/local/opt` so that the default HomeBrew `tcl-tk` and the special one for `xschem` can coexist.

And thanks for your great work on `xschem`!

StefanSchippers, 2021-09-26:
I remember the other big sur user had brew packages installed somewhere in /opt/ or /opt/X11 or somewhere else, and he built the xquartz tcl-tk in /usr/local/opt. But if brew packages are already in /usr/local/opt then ok i will specify to check paths to avoid collisions.