StefanSchippers/xschem #25: no true/useful batch netlist behavior: need option to not fork, not netlist in background & return accurate exit-status

sto6, 2021-10-04:
using V2.9.9, and attempting 100% batch netlisting in a script, something like:

`xschem -r -b -x --rcfile xschemrc -s -o my.spice -n my.sch </dev/null`

After command exits with status=0, the netlist is not present. Meanwhile there's a background xschem still running, and a while later that process terminates and the netlist is available.

IMO this is not proper/useful/standard batch netlisting behavior. For a script that intends to immediately do basic error checks that a valid netlist was produced and do something with it, what can it do? The time-to-wait will depend the complexity of the schematic being processed. I also doubt the current exit-status is meaningful, meaning netlisting in a background process HIDES (I suspect?) any useful exit-status from the script (that of the background worker that does the actual netlisting).

My preference is that batch just works: that non-background mode becomes the new default without any new options required. Is there any user visible benefit/purpose to continued background batch netlisting? Else at least add an option to not-fork, to stay in foreground, or (if absolutely necessary to stay in background): internally to wait on the subprocess & do not exit until the netlist is actually present & complete, then also propagate the accurate exit-status.

off-topic: I'm using the 2.9.9 source tar download from the sourceforge site (as it was at approx. Aug 29 2021). When I look at the 2.9.9 tag in the github, it dates from Feb 10 2021: that seems about a month too old. The sourceforge download I used has files from March. Are these versions in-sync? Which 2.9.9 release should I use? Thanks.

StefanSchippers, 2021-10-05:
Hi! i am not at home, so will follow up in more detail this evening. Add -q
option and xschem will exit after netlisting. do not use -b option. netlist
should be in netlist directory, usually ~/.xschem/simulations. You can
change the directory though.

StefanSchippers, 2021-10-05:
This is the correct command:
` xschem -r -x -s -q -n -o . --netlist_filename my.spice ~/.xschem/xschem_library/test_ac.sch`
where:
`-o . `: set output netlist path to current directory
`--netlist_filename my.spice `: set name of netlist file. If omittet it uses the schematic name with .spice extension.
_The reason there is a `-o` option for path and a `--netlist_filename` option for netlist name is because xschem has also a split netlisting mode, where one file per schematic is produced. In this case you only can specify the output path with -o._

StefanSchippers, 2021-10-05:
For the sourceforge / github issue:
2.9.9 tar file was added late in sourceforge, i forgot to upload the file, however i checked out 2.9.9 when making the tar file.
Please note there is a difference between github and sourceforge. The sourceforge distribution has no 'configure' mechanism, you just enter the 'src' directory, edit the Makefile, adjust library versions (notably tcl/tk) and type make and make install. 
The github distribution has an automatic configurator; typically you run ./configure --prefix=/where/to/install, followed by make and make install. However there no other differences. Anytime i commit a change in github this change is propagated immediately to sourceforge. This means all src/, doc/, xschem_library files are identical.