StefanSchippers/xschem #26: Feature request: File->New Schematic and File->New Symbol to open new window

hpretl, 2021-10-08:
I think it would be handy if the "New ..." function would open up a new window and leave the current one intact, as often a new schematic is started by copy-pasting stuff from the other window. Similar for symbol creation, usually you need the schematic side-by-side.

StefanSchippers, 2021-10-08:
I have committed the requested addition. Instead of changing the existing 'New Schematic' and 'New Symbol' i have added 2 new menu entries, "New empty Schematic window' and 'New empty Symbol window'.

The main reason is to avoid changing behaviour of existing menu commands. Moreover on some architectures (MacOS) spawning a second xschem window does not work, so the old 'New Schematic/Symbol' can be used as a fallback option.

If you have a chance to test this feature and if it works for you we can close the issue.

Tip: if you select a component and hit 'Alt-e' you will get a new xschem window editing the schematic of the component.
If you press 'Alt-i' you will be editing the component symbol in a new window. In some cases these commands may be useful.

hpretl, 2021-10-09:
Works as intended. Thanks!

closed 2021-10-09 by hpretl