StefanSchippers/xschem #28: "No rule to make target 'src//gui/find_gtk4.o'"

andrsmllr, 2021-11-06:
Hi,

When I try to build xschem from master b62f8b57f720bb7c8402276ad875176a14144326 the initial configure step fails with the following error message:

> $ ./configure
cc -DGENCALL -DRUNTIME -g  -DPLUGIN_SCRIPTS   -Isrc//tmpasm  -DPLUGIN_PARSGEN    -DPLUGIN_GUI     -Isrc//default   -c -o hooks.o hooks.c
...
cc -DGENCALL -DRUNTIME -g  -DPLUGIN_SCRIPTS   -Isrc//tmpasm  -DPLUGIN_PARSGEN    -DPLUGIN_GUI     -Isrc//default -c src//gui/find_gtk3.c -o src//gui/find_gtk3.o
make: *** No rule to make target 'src//gui/find_gtk4.o', needed by 'configure'.  Stop.
./configure: 4: ./configure: ./configure: not found

I tried to work around this error by removing references to gtk4 from `scconfig/src/gui/gui.c` and `scconfig/src/gui/Makefile.plugin` by commenting out a few lines.
With these changes the configure and build steps finish successfully. But running xschem throws the following error message:

> $ ./xschem
Tcl_AppInit() error: can not execute /opt/xschem/share/xschem/xschem.tcl, please fix:
invalid command name "image"
tcleval(): evaluation of script: wm withdraw . failed
tcleval(): evaluation of script: tk_messageBox -icon error -type ok -message        {Tcl_AppInit() err 1: can not execute /opt/xschem/share/xschem/xschem.tcl, please fix:
 invalid command name "image"} failed

I am unsure if the configuration is missing something or the master branch is not stable atm.

giljerard, 2021-11-06:
Same problem here.

StefanSchippers, 2021-11-06:
Thank you for the reports, 
configure is now fixed, the problem was in upstream configure (scconfig) code, two gtk4 detection files were added and missing in xschem repo. I have added the missing 2 files.
do a 'git pull' and try again. Let me know if you have further problems.

Thank you

StefanSchippers, 2021-11-06:
There is a compiler warning about a file in scconfig/src/gui/gui.c (pointer type mismatch) 
I have verified that this is not causing any problem. It is however notified upstream to the scconfig developer so will soon be fixed. (I don't like compiler warnings). No problem and no effect on xschem build.

StefanSchippers, 2021-11-06:
Above warning fixed.

andrsmllr, 2021-11-06:
Thanks @StefanSchippers for the quick reaction. I can verify that the issue with configure is fixed now.

The second issue still remains however. Building from master 5e006e78f7a4f1951254abb733766f4b0d8c05c8 gives:

> $ /opt/xschem/bin/xschem
Tcl_AppInit() error: can not execute /opt/xschem/share/xschem/xschem.tcl, please fix:
invalid command name "image"
tcleval(): evaluation of script: wm withdraw . failed
tcleval(): evaluation of script: tk_messageBox -icon error -type ok -message        {Tcl_AppInit() err 1: can not execute /opt/xschem/share/xschem/xschem.tcl, please fix:
 invalid command name "image"} failed

StefanSchippers, 2021-11-06:
What is the version of tcl-tk on your system? 
tcl and tk packages usually have the name tcl8.x, tcl8.x-dev, tk8.x, tk8.x-dev

andrsmllr, 2021-11-06:
Tcl/Tk are both at version 8.6. My build is running on Ubuntu 18.04 in WSL2. 
> =====================
Configuration summary
=====================
Compilation:
 CC:        gcc
 debug:     no
 profiling: no
Paths:
 prefix:        /opt/xschem
 user-conf-dir: ~/.xschem
 user-lib-path: ~/.xschem/xschem_library
 sys-lib-path:  /opt/xschem/share/xschem/xschem_library/devices
Libs & features:
 tcl:       -ltcl8.6
 tk:        -ltcl8.6  -ltk8.6
 cairo:     yes
 xrender:   yes
 xcb:       yes
Configuration complete, ready to compile.

With Ubuntu 20.04 - also in WSL2 - I get the same result.
Could the problem be due to WSL2 with X11 forwarding?

StefanSchippers, 2021-11-06:
No problem is some image data (png images encoded in base64) not recognized. I need to figure out why, since i don't have any issue on my system (also with tcl-tk 8.6)

StefanSchippers, 2021-11-06:
It looks like some tcl-tk versions do not recognize png format and i have to convert to the  older gif format

StefanSchippers, 2021-11-06:
can you try in a terminal: 'man 3tk photo' and see what image formats are supported? 
on my system the man page says:
At present, only PNG, GIF and PPM/PGM formats are supported,
while on older tcl-tk versions:
At present, only GIF and PPM/PGM formats are supported

In the mean time i will convert all images to gif. I have edited and add some images, and i thought png was the default, i was mistaken. will fix asap.

StefanSchippers, 2021-11-07:
I have converted all toolbar small images from png to gif, so now on my system xschem works also with tcl-tk 8.4, which is a version released 20 years ago. I hope it now works on your system. Please test. Thank you very much for reporting this issue.

andrsmllr, 2021-11-07:
Okay, I updated to master 9fd56b8f913652483e15529707bf81d3c8cbe54b.
At first the issue was still there on Ubuntu 20.04 (WSL2). Then I realized that the DISPLAY variable was pointing to the IP of the Wifi adapter [1].
After manually switching the DISPLAY variable to the correct IP - i.e. the IP of the adapter named "vEthernet (WSL)" in Windows 10 - xschem works now!

However, doing the same on Ubuntu 18.04 does not resolve the issue. The error message (`...invalid command name "image"...`) is gone, but no window appears after launching xschem. This is very likely another issue related to X11 forwarding in WSL(2) though and not caused by xschem.

Therefore I think the problem regarding `find_gtk4.o` is fixed and the issue can be closed. If I still have troubles with xschem with WSL(2) + X11 I will open a new issue.

Many thanks to you @StefanSchippers for the quick support 👍. Now I can continue watching your xschem tutorial video on YT :-)

---

[1] to determine the DISPLAY variable in WSL2 I had used ` export DISPLAY=$(awk '/nameserver / {print $2; exit}' /etc/resolv.conf 2>/dev/null):0` in my .bashrc, as described in https://stackoverflow.com/questions/61110603/how-to-set-up-working-x11-forwarding-on-wsl2/66645230#66645230 .
This gave me the wrong IP address though. One way to fix this is to manually set DISPLAY to the correct IP address, i.e. the IP address of the adapter named "vEthernet (WSL)").

StefanSchippers, 2021-11-07:
Ok, will close, if you have issues on Ubuntu 18.04/WSL please file another issue. 
Just for testing try to launch a simple xorg application like 'xclock' and see if it works. 
If that one works and xschem does not shhow up then there is some other issue to look at. 
Many thanks

closed 2021-11-07 by StefanSchippers