StefanSchippers/xschem #29: Freezing Xschem Editor Window

rowhit, 2021-11-14:
Hi,

I recently came across the Xschem while exploring skywaters project. It is really a great project but I am unable to use it. I am running ubuntu 20.04 and uses the default xschem provided by the distribution. I am not sure why xschem freezes with few mins of the launch. I am attaching a image showing the issue. I am sure it is related to my graphics chipset issue somehow. Could you please help me debug this, let me know what information to post here?

[image]

StefanSchippers, 2021-11-14:
Hi, the default xschem bundled with the Ubuntu distribution is too old. It is probably older that the skywater project itself. 
Please clone the github source repository on your computer:
`git clone https://github.com/StefanSchippers/xschem.git  xschem_git`

This will create a xschem_git directory where you should build and install the program. Follow the instructions [here](http://repo.hu/projects/xschem/xschem_man/install_xschem.html). You need to install the required packages, notably the Xorg development files, tcl-tk libraries and development files, libXpm, and a few more.

rowhit, 2021-11-14:
Thanks a lot for the response. I believe I had tried this before and since compilation failed I used the distro default. Here the error I get :+1:

rowhit, 2021-11-14:
I have been able to successfully compile and it seems to be working smoothly. I apologise for creating frivolous issue. Since my amdgpu has given grief in the suspected it rather than the xschem. Thanks again for quick response. I believe this issue can be closed. I may reopen if the issue re-appears.

StefanSchippers, 2021-11-14:
Good!  Always good to get user feedback. Thanks

rowhit, 2021-11-14:
It may sound unreasonable what I am noticing is that if I move my cursor away from xschem window and try to come back then window freezes and becomes unresponsive (I am working on mutliple desktops). I have verified it now multiple times. Do you have any suggestions as why xschem may be doing that. I have tried it with multiple xschemrc (including default or none) file now but still same results.

StefanSchippers, 2021-11-14:
- Do you have Xorg or Wayland graphics?
Can you please try these: 
- If you press 'Escape' when the mouse pointer is inside xschem's window does it redraw the window?
- Try to start xschem from a terminal (if you didn't already do so), don't put xschem in the background  (like with 'xschem &')

rowhit, 2021-11-14:
- Xorg or Wayland? I am using Xorg (Wayland cause issues with several other applications in general).
- No, it was not helping.
- I am launching the xschem from terminal but was using & to run in the background. May be this was the issue. I am now launching without & at the end. I will share my experience if the problem recurs.

StefanSchippers, 2021-11-14:
Xschem by default uses the terminal as a command shell. You can run `xschem -b` to detach xschem from the terminal, in this case you can run: `xschem -b &` without any issues.
If you just do `xschem & `it tries to write the prompt but since it is in background application will be stopped. This happens for example if you run a shell like bash with `bash &`.

rowhit, 2021-11-14:
Got it. Thanks for guidance. It seems to work fine now. I probably should go through the xschem documentation more thoroughly.

closed 2021-11-14 by rowhit

StefanSchippers, 2021-11-14:
Thank you for your feedback anyway. I took a note so i will add a specific item in the docs.