StefanSchippers/xschem #30: Depletion NMOS

Sturgeon637, 2021-11-20:
I don't see depletion load nmos transistor.  I'm looking at a design from 1976 and it is using depletion load nmos logic.  Can it be added to xchem. I'm hoping to dump verilog out an run sims.

StefanSchippers, 2021-11-20:
I can add a specific symbol for a depletion NMOS, however a depletion mode NMOS is just a NMOS transistor with negative threshold. For Spice simulations you can use the standard nmos4.sym symbol and change the 'model' attribute to refer to a depletion NMOS.
Anyway for Verilog sims i will add soon a depletion device, makes sense to have one such device in the xschem distribution. 
Thank you for your contribution!.

StefanSchippers, 2021-11-20:
I have added a depletion mode NMOS transistor (`devices/nmos4_depl.sym`) using the verilog `rnmos` primitive, 
Also added a simple testbench simulating an inverter using such transistor, see schematic 
in `logic/test_mos_verilog.sch`, I have simulated it using Icarus verilog, which can be well integrated in xschem,
see [manual section on verilog simulation](http://repo.hu/projects/xschem/xschem_man/simulation.html). Waves can be viewed with gtkwave, a very good open source digital wave viewer.
[image]

Sturgeon637, 2021-11-21:
That was quick!! Thank you!!

StefanSchippers, 2021-11-22:
I will close the issue, if something else is needed on this subject feel free to reopen.

closed 2021-11-22 by StefanSchippers

Sturgeon637, 2021-11-29:
Hello, I'm now trying to use the depletion nmos device and noticed that when a generate a spice netlist the device type is just nmos. Could it be nmos-depl?

StefanSchippers, 2021-11-29:
You need to specify the model name of the depletion transistor. The name of such model depends on the model file. Just select the nmos in xschem, hit 'q' to edit its attributes and change the 'model=... attribute according to the name of the depletion transistor model. Then create the netlist. Depletion nmos transistor is just a nmos transistor with a negative threshold, so it can be used as a current source, like in nmos logic.

Sturgeon637, 2021-12-01:
I get it! Thanks again!

Sturgeon637, 2021-12-01:
How would you handle an enhancement-mode inverter in verilog. This has the pull up nmos gate connected to drain and VCC.  (image below)
[image]

Sturgeon637, 2021-12-01:
I'm thinking I will need an enhancement nmos device for the verilog model to come out correctly. For example nmos4_enh.sym.

StefanSchippers, 2021-12-01:
I have created a `devices/rnmos4.sym` component, this is exactly the same as `devices/nmos4.sym` but in verilog it is simulated as a `rnmos` primitive (resistive nmos). Please ensure you place the enhancement load transistor upside-down (the drain node to the OUT node). I have also updated the test schematic.
the nmos_depl does not need to be placed upside-down since xschem switches source and drain automatically. 
If you prefer to have the rnmos4.sym with the drain node up just let me know i will change the tetlisting rule to switch source and drain when producing verilog netlist.

Sturgeon637, 2021-12-01:
Looking at other documentation, the Enhancement Load has the drain node up.  I say we put it up.
[image]

StefanSchippers, 2021-12-01:
I have changed the rnmos4.sym transistor, so it will be placed drain side up as you showed, xschem verilog netlister will generate the correct primitive to get correct simulation results.

Sturgeon637, 2021-12-02:
Sorry for this question in the thread but why does "show net names on pins" not seem to work? Is it a color issue or a parameter issue? 
[image]

StefanSchippers, 2021-12-02:
There is a double enable mechanism, one is global as you have set in the menu. Next, instances you want to display attached net names should have a '**net_name=true**' attribute set.
You can select all instances together, then press 'q', enable checkbox '**preserve unchanged props**' (this is very important) and add **net_name=true**, it will be added to all selected components without changing their existing attributes.

Sturgeon637, 2021-12-02:
How do I get the net names on the pins of the  Depletion-mode logic cells I created? For example in the NOR_D and BAND_B cells in the image. Thanks again and your tool is awesome.
[image]

StefanSchippers, 2021-12-02:
Oh, yes you need to do a couple more things.
Descend into one of the symbols, i am showing here an inverter, but the concept is the same for other cells
[image]
, so suppose i have an inverter with input A and output Z, descend into it and add some labels as shown in picture: