StefanSchippers/xschem #31: Symbol Properties not Updating

rowhit, 2021-11-21:
I am struggling to figure as why I cannot get the properties in the instantiated version of a symbol are not updating. Here is an example of updated symbol.
[image]

This figure shows that in annotation the new properties are showing up but not in the edit properties dialog. I've done Alt+S (etc for reload). Closed the windows and reopened but still I see the names from the old symbol and properties name.
[image]

rowhit, 2021-11-21:
I figured out if delete instance for which properties are updated and then you save the schematic and re-instantiate this symbol then you get the updated properties. However, if your design is large then it could tedious. Although, the good thing is all the schematic and symbols are easily editable via test thus updating is not difficult via sed or something like that. However, I am curious if there is an inbuilt way to refresh without performing above steps or what I am seeing is a bug?

StefanSchippers, 2021-11-21:
If you change the underlying symbol definition, propagating the changed attributes to the symbol instantiations is very dangerous.
Should all instance attributes that do not match symbol be deleted? this is extremely dangerous, a library  update (or using an old version of it) could break an existing design. If an attribute changes name in a symbol there is no automatic way to recognize how to bind the new attribute to the older one, human intervention is needed.  
However you can block select all components of a given type and change an attribute (or more), change will be applied to all selected objects, see this [video](https://xschem.sourceforge.io/stefan/xschem_man/video_tutorials/edit_attributes.mp4). You can also use the '[search](https://xschem.sourceforge.io/stefan/xschem_man/video_tutorials/search_replace.mp4)' function to locate and select the objects needing this kind of update

rowhit, 2021-11-21:
So, this is intended behaviour. Thanks for the response. I believe some of the questions I have are not really issues but rather limited understanding of the tool which will only come with experience (more I use it more I understand).

closed 2021-11-21 by rowhit

StefanSchippers, 2021-11-21:
In some cases i also had situations like these, and as you mentioned doing a replacement with sed is also a possibility.
Considering that in xschem files a component instantiation looks like:
C {nmos4.sym} 550 -160 0 0 {name=m1 model=cmosn w=5u l=2u m=1  net_name=true}

changing an attribute (example: l to length) can be done with:
`sed -i '/^C/s/ l=/ length=/'  <list of .sch files>`
If the instantiation line is split on more lines the replacement could be trickier, however if your attribute names are sufficiently long to be unique replacing with sed shoud be safe. I remember when working with Cadence we had the same issues when library symbols were changed, leaving dangling attributes and wrong netlist. No automatic flow was existing to fix the design. 
I am open to suggestions how to handle these situations.

StefanSchippers, 2021-11-21:
may be a menu option to re-sync instance attributes to symbol definiton should be added, this could add new symbol attributes (with no assigned value) to the component instances, however you will end up with components having old and new attributes and a human must assign the correct values to new attrs and delete teh old ones. In essence it is a complex task.

rowhit, 2021-11-21:
I understand now. I usually long names for readability, clear annotations in my schematic. Thus conflicts are rare. 

Just a though, may be a menu item that allows you to dump a log with warning of instances where a symbol or its attribute have been updated can be useful. This would atleast permit user to choose what they intend to update.

StefanSchippers, 2021-11-21:
I have taken a note, looks an interesting addition to be done. Thank you for now!