StefanSchippers/xschem #36: Possible SIGSEGV in xschem on startup

ericsmi, 2021-12-23:
I am getting the following error upon starting xschem when it is built by cloning the GitHub repo.  This xschem was built with only the --debug option given to configure.   The issue is repeatable on my machine. 

(gdb) run
Starting program: ~/tools/xschem/src/xschem 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff6811700 (LWP 19367)]

Thread 1 "xschem" received signal SIGSEGV, Segmentation fault.
0x00005555555c1dfb in Tcl_AppInit (inter=0x55555577a1b0) at xinit.c:1681
1681	    for (i = 0; i < formats_reply->num_formats; i++) {

$ cat /etc/os-release
NAME="Ubuntu"
VERSION="20.04.2 LTS (Focal Fossa)"
$ echo 'puts $tcl_version;exit 0' | tclsh
8.6

I am mostly sure I installed TCL/TK via "sudo apt install tcl-dev tk-dev -y"  I have built magic from source and it seems to work on this machine so I think the TCL install is likely ok.

StefanSchippers, 2021-12-24:
Hi, i am not at home now.
Anyway, after doing  ./configure ... edit config.h and comment out (/* ...
*/) definitions for HAS_XCB and HAS_XRENDER. There are no visible
performance issues by doing so.
Try to rebuild and run. If you can please let me know if this solves the
issue.
Thanks, best wishes.
Stefan

StefanSchippers, 2021-12-24:
Also note that xschem runs on a X11 display server (usually Xorg). On
Xwayland may be commenting the definitions as explained before may work.
Stefan

ericsmi, 2021-12-24:
Commenting out HAS_XCB and HAS_XRENDER worked and now xschem seems to run without issue.  I was able to load all the SkyWater130 stuff and draw some simple things.  Does this mean there is an issue with XCB on my machine?  Or maybe XCB is incompatible with being run from vncserver.   Anyway, it seems like xschem is ok.

StefanSchippers, 2021-12-24:
yes, maybe the vncserver does not offer that functionality but the
configure detected xcb on the host. may be i could catch the error when
setting up xcb connextion and revert automatically to xlib runtime in case
of problems.
Thank you for pointing out the issue.
Merry Xmas.

closed 2021-12-27 by StefanSchippers