StefanSchippers/xschem #37: Create LVS netlist from command line

hpretl, 2021-12-27:
I would like to create an LVS netlist from the command line. Currently I use

`xschem -n -s -q --no_x $CELL_SCH -N $NETLIST_SCH`

to write the netlist, but I would like to have the `.subckt` in the file? Is there an option to do this? In the GUI there is the "LVS netlist" option.

StefanSchippers, 2021-12-27:
Yes this can be done. There is no specific command line option, but there is a tcl variable, and it can be set with --tcl:
`xschem -n -s -q --no_x  --tcl 'set top_subckt 1'  $CELL_SCH -N $NETLIST_SCH`

The --tcl '...' can be used to change (almost) any variable that you have in the xschemrc file.

hpretl, 2021-12-27:
Perfect, works. Thanks!

closed 2021-12-27 by hpretl