StefanSchippers/xschem #38: Read-in of Verilog netlist

hpretl, 2021-12-28:
Is there a possibility to read-in (or convert) a (powered) Verilog netlist? (similar to `make_sch_from_vhdl.awk`?)

Background: I am working towards an analog-on-top workflow for SKY130, and my line of thinking is that everything analog is anyway done in `xschem`, so when I could include the digital part as well as I can build the top level and write the full netlist from `xschem`.

StefanSchippers, 2021-12-28:
Hi, is the verilog file a circuit with standard cells or is it a more
complex design with more hierarchy levels?
Regards
Stef

hpretl, 2021-12-28:
Just one level with Standard cells. Essentially the result of a synthesis run.

StefanSchippers, 2021-12-28:
i have a script, if you can send me a test netlist (a cut down with a few
stdcells) i can do some tests.
Another approach is saving the synthetized design in spice format. Many
synthesis tools allow that.

hpretl, 2021-12-28:
I want to be able to just use open source, so I am stuck with OpenLane. Here is a simple test case (counter). This is the powered Verilog that I get at the end from OpenLane. If I could read this in `xschem`, then I can embed it into a larger design.

[counter.v.gz](https://github.com/StefanSchippers/xschem/files/7786373/counter.v.gz)

StefanSchippers, 2021-12-28:
Ok, Thank you, will work on this. 
Please consider also that you can embed just the verilog code into a xschem schematic, you just draw the symbol and a schematic with the interface pins only (input/inout, output) and load the verilog code into the schematic. See this video tutorial for various ways to simulate rtl/behavioral blocks:
https://xschem.sourceforge.io/stefan/xschem_man/video_tutorials/multiple_schematics_bound_to_symbol.mp4

StefanSchippers, 2021-12-31:
I have some progress on this issue, tranlating your counter.v example,
need to cleanup a bit the import script, then i will tell you the procedure.
[image]

hpretl, 2021-12-31:
Great! I think this will be quite useful for SKY130/OpenLane users. BTW, I am working on making useful add-ons to SKY130, maybe interesting for you (https://github.com/hpretl/iic-osic).

StefanSchippers, 2021-12-31:
Very interesting! for

 mixed analog/digital designs i am also adding new features to xschem.
One of these is the ability to display waveforms from ngspice dimulations (later also from Xyce), and one nice feature is to be able to display bussed signals directly from analog spice sims. I haven't seen good analog viewers that are able to do that. Ability to stack digital waves and analog waves is very useful when debugging.

[image]

hpretl, 2021-12-31:
The embedding of simulation waveforms is quite useful, I will look into it! For powerful simulation data crunching, I am gearing towards using Python right away, mainly due to lack of decent waveform viewers for analog incl. a calculator-like feature (`gtkwave` is OK for digital). I am thinking about https://github.com/gmagno/spyci as my base.

StefanSchippers, 2022-01-02:
Hi, i have committed the [verilog import](https://github.com/StefanSchippers/xschem_sky130/blob/main/xschem_verilog_import/README.md) on your `counter.v` example.  Can you please take a look and see if it works on your side?

hpretl, 2022-01-06:
Thanks! I have tested the script on a more elaborate example, unfortunately, it throws an error:

```
 process_subckt() : audiodac--> 
clk_i ds_n_o ds_o fifo_ack_o fifo_empty_o fifo_full_o fifo_rdy_i mode_i rst_n_i fifo_i[15] fifo_i[14] fifo_i[13] fifo_i[12] fifo_i[11] fifo_i[10] fifo_i[9] fifo_i[8] fifo_i[7] fifo_i[6] fifo_i[5] fifo_i[4] fifo_i[3] fifo_i[2] fifo_i[1] fifo_i[0] osr_i[1] osr_i[0] 
*.PININFO clk_i:I ds_n_o:O ds_o:O fifo_ack_o:O fifo_empty_o:O fifo_full_o:O fifo_rdy_i:I mode_i:I rst_n_i:I fifo_i[15]:I fifo_i[14]:I fifo_i[13]:I fifo_i[12]:I fifo_i[11]:I fifo_i[10]:I fifo_i[9]:I fifo_i[8]:I fifo_i[7]:I fifo_i[6]:I fifo_i[5]:I fifo_i[4]:I fifo_i[3]:I fifo_i[2]:I fifo_i[1]:I fifo_i[0]:I : osr_i[1]:I osr_i[0]:I
ERROR in .PININFO:  : --> *.PININFO clk_i:I ds_n_o:O ds_o:O fifo_ack_o:O fifo_empty_o:O fifo_full_o:O fifo_rdy_i:I mode_i:I rst_n_i:I fifo_i[15]:I fifo_i[14]:I fifo_i[13]:I fifo_i[12]:I fifo_i[11]:I fifo_i[10]:I fifo_i[9]:I fifo_i[8]:I fifo_i[7]:I fifo_i[6]:I fifo_i[5]:I fifo_i[4]:I fifo_i[3]:I fifo_i[2]:I fifo_i[1]:I fifo_i[0]:I : osr_i[1]:I osr_i[0]:I
```

Here is the Verilog file I tested:

[audiodac.v.gz](https://github.com/StefanSchippers/xschem/files/7820710/audiodac.v.gz)

Since the script is targetted at SKY130 I also propose the following change:

```
 #xschem_lib_path="$HOME/.xschem/xschem_library/xschem_sky130/sky130_stdcells"
 xschem_lib_path="$PDK_ROOT/sky130A/libs.tech/xschem/sky130_stdcells"
```

StefanSchippers, 2022-01-06:
I have done a fix and it seems now ok importing the audiodac.v.
Can you please test? 
Thanks!

hpretl, 2022-01-06:
I tested the updated script, and it produces a schematic and symbol. However, when I use a netlist generated from this schematic view in an LVS the LVS fails, and I don't know why (could be many things: generated SCH wrong, SKY130 SC lib SPICE models wrong, LVS just gets lost, ...). I will need some time to debug this on a simpler example that I currently have, and I will get back to you.

When I extract a SPICE netlist from the generated SCH I get these errors. Are they critical?

```
tcleval(): evaluation of script: wm deiconify .infotext failed
         : invalid command name "wm"
```

I use this to create the SPICE netlist:

```
xschem -n -s -q --no_x --tcl 'set top_subckt 1' "$CELL_SCH" -N "test_$NETLIST_SCH"
```

StefanSchippers, 2022-01-06:
I have updated the import script. It is now doing much better, and xschem produces a spice netlist without warning / errors.
However synthesis tools produce very fancy names for nets, i believe some are not allowed in spice so we need to define some translation rules. Example: 
`wire \dsmod.accu1[15]`

in spice the '.' is a hierarchy separator, i don't know what about for backslashes. I have now set up the translator to generate a schematic with net names that result in identical names in spice netlist produced by xschem. However I think we may drop the backslash and convert the dot to something different. What do you think?.

Side note: i have also fixed the infowindow error when netlisting with xschem. Thank you for pointing out the issue.

StefanSchippers, 2022-01-06:
One more problem i see is with kind of 'two' dimensional vector nets, like: `\fifo.fifo_reg[14][0]`
Xschem (and Verilog itself) has the concept of vector nets, for example `net1[7:0]` expands to `net1[7], ..., net1[0]`. When such nets are founds the importer script tries to glue single nets found in gate level netlist into a vector port, this is why in audiodac.sch you see in/out port declared as vectors where needed (example: fifo_i[15:0]).
I need to figure out what to do when such '2 dimensional nets are found. Verilog uses the backslash to ignore any special meaning of brackets and take the net name literally as is.

hpretl, 2022-01-07:
> in spice the '.' is a hierarchy separator, i don't know what about for backslashes and the '$' is i believe an end of line comment in spice. I have now set up the translator to generate a schematic with net names that result in identical names in spice netlist produced by xschem. However I think we may drop the backslash and convert the dot and '$' to something different. What do you think?.

I think it would be fine to convert all unusual characters to `_`, or remove them completely.

The main usage of the schematic after conversion that I see is

1. Run a digital simulation on transistor level for e.g. sign-off or mixed-signal.
2. Run an LVS on transistor level.

In both cases, no one would interpret the internal schematic signal's name. Only during debug session, someone would want to compare Verilog net names and schematic net names. As long as they are "close enough" I guess that would work.

However, interface names (as in the symbol) should match between Verilog and schematic, but if I understand you correctly these are not causing issues.

> I need to figure out what to do when such '2 dimensional nets are found. Verilog uses the backslash to ignore any special meaning of brackets and take the net name literally as is.

I think the same applies here. So `\fifo.fifo_reg[14][0]` could be translated to `fifo_fifo_reg_14_0` in the schematic.

hpretl, 2022-01-07:
I did a bit of debugging why the LVS with `netgen` fails (using latest version of `xschem` and `make...`:

`netgen` says: "Instance of sky130_fd_sc_hd__inv_2 has only 5 of 6 ports"

This could be due to the net name format. The Magic extract looks like this:

```
X_1515_ _2127_/Q VGND VPWR _1515_/Y VGND VPWR sky130_fd_sc_hd__inv_2
```

Your conversion tool has net names like this:

```
X_1515_ \fifo.fifo_reg[15] VGND VGND VPWR VPWR _0528_ sky130_fd_sc_hd__inv_2
```

1. The backslash seems to be OK, but maybe the `.` or the `[]` cause issues?
2. Also, the VGND and VPWR seems to be a bit mixed up, not sure why this is and what is correct.

For the LVS I have to add the SPICE netlists of the SC by doing a

```
cat /usr/local/share/pdk/sky130A/libs.ref/sky130_fd_sc_hd/spice/sky130_fd_sc_hd.spice >> audiodac.sch.spice
```

StefanSchippers, 2022-01-07:
Thank you for the info. 
About the pin ordering xschem aims to reproduce the same order as it appears in the spice netlist in: 
`.../share/pdk/sky130A/libs.ref/sky130_fd_sc_hd/spice/sky130_fd_sc_hd.spice`,
taking as an example the inv_2 cell:
```
.subckt sky130_fd_sc_hd__clkinv_2 A VGND VNB VPB VPWR Y
X0 Y A VPWR VPB sky130_fd_pr__pfet_01v8_hvt w=1e+06u l=150000u
X1 VPWR A Y VPB sky130_fd_pr__pfet_01v8_hvt w=1e+06u l=150000u
X2 Y A VPWR VPB sky130_fd_pr__pfet_01v8_hvt w=1e+06u l=150000u
X3 VGND A Y VNB sky130_fd_pr__nfet_01v8 w=420000u l=150000u
X4 Y A VGND VNB sky130_fd_pr__nfet_01v8 w=420000u l=150000u
.ends
```

so xschem generated netlist seems correct ( Input(s), ground, pwell tap, nwell tap, supply, output(s)).
The same ordering is also used in cdl netlists.
Will do some experiments by replacing strange characters ('.', '$') with alternatives. I don't think the '[' and ']' will cause trouble for spice/netgen LVS, however if they do i can replace these as well.

StefanSchippers, 2022-01-07:
I have updated the verilog importer script to rename 'offending' node names, leaving the interface ports of the module unchanged.
I don't see any warnings when producing the spice netlist from xschem, this means that all nets in the design are driven and there are no shorts or multiple drivers on a net.
I have cleaned up the schematic (removing filler, taps and decaps) and saved into 'audiodac_clean.v', just to do a very basic logic smoke test:
[image]

I have also added the logic functions to all used std cells in the design (combinatorial & flops), so i can verify that after issuing a reset and a clock pulse all nets in the design are at a well defined logic state (no 'X' states). 
This can be done as shown: 
1)select all input ports and press the '0' key to set all these nets to logic 0 value:
[image]
2) select only the clock pin and press '1' (to set clock high)
[image]
as you see all outputs are no more cyan (cyan is the 'X' logic state, for 'unknown') but are either red (1) or blue (0). After playing a bit toggling the clock (the '4' key on a selected net or group of nets toggles logic state) I see signals moving in the circuit, tho i don't know the exact circuit function. The good thing is i don't see 'X' anywhere.

[image]

StefanSchippers, 2022-01-07:
Just a hint when working with large schematics:
If you need to find a net name ('lab' attribute) or an instance name ('name' attribute) you can use the search function (ctrl-f): 
[image]
After doing the search the found elements are highlighted, however in a big design it is still difficult to locate the search:use the '5' toggle key to hide everything but the highlight:
[image]
You will see the elements and can easily zoom to them (press '5' to display the circuit again).
Another solution is to use the search function and check the 'select' checkbox. After selecting, a zoom full ('f') will zoom to the right place. As before the '5' key hides everything but the selection.

hpretl, 2022-01-08:
Thanks for all the hints and work on this thread!

I did some more tests on the failing LVS issue (LVS is my goal, because if this works, then I know the generated schematic is correct). I switched to use `spm`, which is an example contained in the `OpenLane` distribution.

What I realized:

- The LVS reports 392 instances in all cases for `spm`, either layout extract, schematic or Verilog (after merging of parallel devices, like the filler cells). 
- However, the number of nets is different: There are 427 nets in the layout extract and the Verilog, but only 367 nets in the SPICE netlist generated from the schematic.

Any idea why the number of nets might be different after the Verilog-to-SCH conversion?

Here is the complete folder attached of my experiment.
[test_lvs.tar.gz](https://github.com/StefanSchippers/xschem/files/7833379/test_lvs.tar.gz)

StefanSchippers, 2022-01-09:
I have made a fix in the import script and now xschem spice netlist also has 427 nodes.
The problem was due to nodes named like:
` \genblk1[2].csa.sc` that were not corrected due to a bug, and xschem gets confused and truncates the part after the [2].
These nodes are now translated to `genblk1_2__csa_y`. 
I will review the translation procedure, to avoid name clashes,
if one node is `aaa[2]bbb` and another is `aaa.2.bbb` they are now translated to the same node `aaa_2_bbb`.
However for the examples tested so far this does not happen.

hpretl, 2022-01-09:
Proposal for a naming conversion that avoids potential name clashes:

- A `\` is just removed.
- A `[` or `]` is converted to `_`
- A `.` is converted to `__`

Regarding the issue with the number of nets: We are almost there, LVS is matching, just complains about the pin list. :-)

We need the power supply pins (VPWR, VGND) in the subcircuit definition, and thus also in the symbol. They are already used in the netlist properly:

```
XFILLER_6_52 VGND VGND VPWR VPWR sky130_fd_sc_hd__fill_1
```

but not in the subcircuit def:

```
.subckt spm clk p rst y x[31] x[30] x[29] x[28] x[27] x[26] x[25] x[24] x[23] x[22] x[21] x[20]
+ x[19] x[18] x[17] x[16] x[15] x[14] x[13] x[12] x[11] x[10] x[9] x[8] x[7] x[6] x[5] x[4] x[3] x[2] x[1]
+ x[0]
```

Here is the report from the LVS on the cell `spm`:
[comp.out.gz](https://github.com/StefanSchippers/xschem/files/7834738/comp.out.gz)

StefanSchippers, 2022-01-09:
I have added a flag variable `dis_implicit_subckt` that keeps implicit connections (power pins) in the .subckt declaration.

For net renaming the '_' can be used as an 'escape' character and as such it must escape itself:
The following might work:
```
_ translated to __
\ is just removed (I believe it is safe since it is already an 
                   escape for verilog, if not we can translate to _b)
. translated to _d   (d for 'dot')
[ translated to _o   (o for open bracket)
] translated to _c   (c for close bracket)
$ translated to _s   (s because it is similar to $)
```

hpretl, 2022-01-09:
Works!! :-)

The LVS matches, tested on `spm` and `audiodac`.

I used this LVS script: https://github.com/hpretl/iic-osic/blob/main/iic-lvs.sh

I guess we should close this issue then? Thanks for the support!

StefanSchippers, 2022-01-09:
By experience I think some testing is needed on different designs. Import
script is not a true Verilog language parser and depends on the netlist
formatting. If the synthesis tools are consistent the verilog import should
work. If some differences are introduced I must handle the various variants
in the script. I an thinking in particular of some other strange net names.

This is also the reason such a verilog import can not be advertised as a
general purpose "push button" tool. It must be tuned on a specific
toolchain flow. A true generic verilog importer is a complex project (more
than xschem combined), may be it can be done by borrowing the verilog
parser frontend from an existing tool, like Icarus Verilog or Verilator,
but for sure not an easy thing.

If you have a chance to do some more testing that would be great.
Thank you.

hpretl, 2022-01-09:
I will definitely keep using it. If I find issues, I will report them back. So far, the combination SKY130 and OpenLane seems to work fine, and this is my focus at the moment.

StefanSchippers, 2022-01-11:
closed for now, will reopen in case of problem with other test cases.

closed 2022-01-11 by StefanSchippers