StefanSchippers/xschem #39: Request to support to import Viewdraw schematic

lehaianh1986, 2021-12-28:
Hi Stefan,

Can you support to import Viewdraw schematic into xschem? I saw some converter can be modified to reuse such as https://github.com/ljmljz/schematic-file-converter

StefanSchippers, 2021-12-28:
Hi, creating a converter can be done. The difficult part is to get the specification of the file format.
I will look into above link and follow up here.
Thank you

lehaianh1986, 2021-12-29:
Hi Stefan,

I have this document https://drive.google.com/file/d/1URzoXz9L92Im5TuJ1_kPXkHP786zV3_x/view?usp=sharing

lehaianh1986, 2021-12-29:
Hi Stefan,

If you need some Viewdraw file to test, I can send it. Please pm to my gmail lehaianh1986

StefanSchippers, 2022-01-11:
I have started to parse the viewdraw example files, i am starting with the symbols. So far it seems that the translated symbols are slightly different from the images (different text positions, different aspect ratio), and since there is a 1:1 mapping between the viewdraw files and the translated xschem files, are you sure the NMOS.1 and PMOS.1 files you sent me are exactly the ones referring to the images in pictures?

Here is the PMOS.1 converted to xschem PMOS.sym (top is image, bottom is xschem) Some details are yet to be implemented, like the circle color, but these are details, i am concerned about the size for now:

[image]

and here the NMOS.1:

[image]

In your pictures the W=... and L=.. texts are in the same positions in PMOS and NMOS, while the NMOS.1 and PMOS.1 viewdraw files have different coordinates and thus xschem PMOS.sym and NMOS.sym show these texts in different positions.

StefanSchippers, 2022-01-12:
I have some progress in the import process. The following is what i get so far for the dti_28hc_6p5t_40_nand2x1.1 nand schematic:

[image]

However this schematic includes a symbol called **NM.1** for the nmos transistor and **PM.1** for the pmos transistor.
The test case i got from you has **NMOS.1** and **PMOS.1**, and as shown they are of different size. I have put the two transistors manually in the xschem schematic to show they don't fit into the rest of the schematic.

Can you please send me the **NM.1** and **PM.1** files?

Thanks

StefanSchippers, 2022-01-12:
I have verified that the flip flop, dti_28hc_6p5t_40_ffqx1.1, uses the PMOS.1 and NMOS.1 symbols so i am able to generate a valid schematic. If you have a picture of this schematic that would be great so i can adjust labels and all the remaining details.
[image]

closed 2022-03-22 by lehaianh1986