StefanSchippers/xschem #40: Draw schematic GUI got error frequently.

lehoanghiep87, 2021-12-30:
Dear Stefan,
When I open xschem GUI to draw a schematic, when I insert wire I frequently meet an error:
"% tcleval(): evaluation of script: context_menu failed"

After this I can't make any modify to the schematic but can save , close the GUI and need open it again.

Here is picture when I try make a crash happened. If the wire is horizontal or vertical only seem the error less happened.
[image]

Would you please debug it ? If you need more information please let me know.

StefanSchippers, 2021-12-30:
I have a some of questions:

- if you open an empty xschem window and inside the drawing area you press the right mouse button does a context menu near the mouse pointer appear?

[image]
- If you launch xschem from a terminal, at the xscchem terminal prompt, type:
puts $tcl_version
what number do you get ? 
- Usually to place a wire i position the mouse to the first point, then press 'w' and drag the mouse to the end point. Clicking the left mouse button ends operation. Is this your procedure?

Thank you
Stefan

lehoanghiep87, 2022-01-01:
Wish you a happy new year stefan.
1. When I press the right mouse the context menu did appear.
2. When I put the $tcl_version
Here is the warning: 
% $tcl_version
invalid command name "8.5"
3. I use insert wire button at GUI to draw then now I notice maybe when in draw wire mode , I mis-click the right mouse, it make the conflict with context menu then make the error that I reported.
4. I will try testing more to see if I am not mis-click the right mouse the crash error happened or not.

StefanSchippers, 2022-01-01:
Thank you, happy new year to you too. 
I am trying to reproduce the error, but so far i did not get any.
I have tcl8.6 but i am also using a xschem built with an old tcl8.4 and both do not report any error.
If you know the mouse sequence that causes the error let me know.

lehoanghiep87, 2022-01-02:
Dear Stefan,
I did test some different ways, the context error crash seem not happened because of the wire drawing.
It happen if I click right mouse to open context menu then move the mouse pointer before the context menu appear , then you will see this error.
I meet this error because I have bad behavior use both mouse left/right frequently and move mouse around even if I don't use context menu function.

StefanSchippers, 2022-01-02:
Oh Yes! Thank you very much, i am able to reproduce it here too. Ok, now i can work for a fix.
Thank you very much. Will get you informed when done.

StefanSchippers, 2022-01-03:
Hi, in latest xschem commit i have fixed the error you reported. The problem was a race condition where the context menu got closed (by the mouse exiting it) before it completed displaying. 

I have done 2 things:
1) if this error happens just ignore the context menu and do nothing, the program will no more be stuck, everything will be working fine.
2) i have also inserted a check where if the context menu has been already closed just return without trying to access a window that does no more exist.

In my tests i was no more able to trigger the `tcleval(): evaluation of script: context_menu failed` error.
Can you please test ? 
Thank you for reporting this nasty bug!

lehoanghiep87, 2022-01-03:
Dear Stefan,
After update the latest version. I did test and don't meet the context menu again. The GUI become smooth. 
Thank you to resolve it.
Hiep.

closed 2022-01-03 by StefanSchippers