StefanSchippers/xschem #41: Generate hierarchy PDF with hyperlink

lehaianh1986, 2021-12-30:
Can you support to add hyperlink from top schematic PDF page to each sub schematic page when we click over these symbol? And add the name of schematic into each PDF page? Thank you.

StefanSchippers, 2021-12-30:
For hyperlinks i need to investigate. Xschem generates postscript and then postscript is converted to pdf by ghostscript or similar tool.

For schematic name i usually always put the title symbol in every schematic. You can customize your title (title.sym) symbol with the information you like and may be a logo.

This symbol also displays the author name and the time of last modify of the circuit.

[image]

lehaianh1986, 2021-12-30:
Thank Stefan. Yes I can use title symbol to distinguish sub schematic in PDF. Hyperlink or bookmark feature will help a lot with complex schematic having many hierarchy levels.

StefanSchippers, 2021-12-30:
Hi, i have made a commit where a small page title is printed on each pdf page.

[image]

StefanSchippers, 2021-12-30:
Hi, I have implemented in last commit the requested feature for hierarchy pdf export. 
All symbols when clicked will jump to the correspondent schematic.
Can you please test if it works on your side?

lehaianh1986, 2021-12-31:
I meet this error when try to gen PDF even with or without adding title.sym into my schematic.

% problems converting postscript to pdf: Error: /undefined in data
Operand stack:
   --nostringval--   Dest
Execution stack:
   %interp_exit   .runexec2   --nostringval--   --nostringval--   --nostringval--   2   %stopped_push   --nostringval--   --nostringval--   --nostringval--   false   1   %stopped_push   2047   1   3   %oparray_pop   2046   1   3   %oparray_pop   2026   1   3   %oparray_pop   1884   1   3   %oparray_pop   --nostringval--   %errorexec_pop   .runexec2   --nostringval--   --nostringval--   --nostringval--   2   %stopped_push
Dictionary stack:
   --dict:952/1684(ro)(G)--   --dict:0/20(G)--   --dict:99/200(L)--
Current allocation mode is local
GPL Ghostscript 9.25: Unrecoverable error, exit code 1

StefanSchippers, 2021-12-31:
Can you please export the circuit that causes the problem and give it a name with a '.ps' extension instead of default '.pdf' ? This way xschem saves to postscript, then if the circuit is not a company IP or protected
can you send the .ps file to me ? i will investigate. Thanks!

lehaianh1986, 2021-12-31:
Hi Stefan,

You can test this example

[xschem.zip](https://github.com/StefanSchippers/xschem/files/7795968/xschem.zip)

StefanSchippers, 2021-12-31:
Thank you, i found my side a couple of issues and fixing right now, then will test ps->pdf conversion on your example. Thank you for helping me !

StefanSchippers, 2021-12-31:
Ok i have committed the fix.  the problem is that pdf wants plain file names for links, with no directory prefixes. I have fixed that . Can you please update and test again?

lehaianh1986, 2021-12-31:
Thank Stefan, it works perfectly for me.

closed 2021-12-31 by lehaianh1986

StefanSchippers, 2021-12-31:
Just to inform you that i have made another small fix, where symbols that do NOT have a schematic (for example library symbols or spice primitives) do not get a link added (since there is no point to go).