StefanSchippers/xschem #42: Request to support window tab

vietnh89, 2022-01-04:


StefanSchippers, 2022-01-05:
I think this can be done, i add this to the todo list. I have some tasks right now so it will take some days to have this feature.
Thank you

vietnh89, 2022-01-05:
Thank you so much.

StefanSchippers, 2022-01-09:
Hi, I have added the tabbed interface in xschem.
This can be enabled by `View` menu option `Tabbed interface`.
By enabling it you get a tab bar in xschem area:

[image]

This mode can be enable permanently by setting this variable in your xschemrc file:
`set tabbed_interface 1`.

This interface needs some testing, so any feedback is appreciated! Do not use this immediately for production.

When this interface is enabled the menu entries in the File menu to open secondary windows will be disabled. tabbed interface and floating secondary windows are mutually exclusive.
If you need another window to put side by side to a tabbed running xschem however you always can start another xschem process.

Some notes:

- Pressing the ' + ' button adds a new tab, you can then load a schematic as usual.
- Pressing 'ctrl-q' or 'File->Exit' or closing the window (the 'X' in the corner usually) will close the tab.
- The first tab is the 'master tab'. Closing this tab will exit the program. If there are other tabs with modifications xschem will ask you if you want to close those secondary tabs or not, before closing the master and quitting the program.
- Currently there is no function to move the tabs or change their positions, may be i will add this in the future.
- If you close all additional tabs leaving only the master tab you can switch back to non tabbed interface and the menu options to open secondary windows (File menu) will be enabled again.
- You can select objects in one window tab (by dragging a selection rectangle, for example), then press 'c' or 'm' (for copy or move) and copy/move these objects to another tab window , by going to the tab bar and selecting another window. Copy/paste (ctrl-c, ctrl-v) works as well.
- I have set a limit of 20 tabs, not due to limitations in the program, but simply because 20 buttons will not fit in the tab bar (unless a very big window and a very capable monitor is used). I think this is a reasonable limit.

vietnh89, 2022-01-10:
Hi Stefan,
Thank for your supporting. I will test it soon.

vietnh89, 2022-01-10:
Hi Stefan,
This feature looks fine. I have few suggestion:
1. When i open both sch and sym, there are same name. So i think you can add more extension .sch or .sym to tab title
2. When the cell is opened in 1 tab, it should not be opened in other tab. I think you can set active window for the tab which opened the cell.
[image]

StefanSchippers, 2022-01-11:
I have pushed some new commits to xschem:
If tabbed interface is enabled, selecting a component and pressing 'Alt-e' will descend into the schematic in another tab,
while pressing 'Alt-i' will descend into the symbol.
(menu Edit->Edit schematic in new window/tab, Edit -> Edit symbol in new window/tab)
If tabbed interface is not enabled new windows will be opened.

If you try to open a file that is already displayed in a window / tab a warning is now printed.

Pressing 'Ctrl-left arrow' and 'Ctrl-Righ arrow' wil allow you to switch between adjacent tabs. using the keyboard.

If you have a chance to test these out, please report back any issue you find.
Thanks
Stef

StefanSchippers, 2022-01-21:
tabbed windows are implemented, so far did not see problems, closing for now...

closed 2022-01-21 by StefanSchippers

vietnh89, 2022-01-21:
Ok. I did not see problem too.
Thanks for support.

slagernate, 2025-08-19:
Hijacking this --would it be possible to have e.g. `ctrl-tab` cycle to the previously active tab? This is common behavior in e.g. web browsers and really makes life great when switching between tabs. Thanks!

StefanSchippers, 2025-08-20:
> Hijacking this --would it be possible to have e.g. `ctrl-tab` cycle to the previously active tab? This is common behavior in e.g. web browsers and really makes life great when switching between tabs. Thanks!

@slagernate this can be done with Ctrl-Left arrow or Ctrl-Right arrow

slagernate, 2025-08-20:
Hmm, Ctrl-Left/Right arrow switches to adjacent left/right tabs, not the previously active tab.

slagernate, 2025-08-21:
I appreciate the timely response and implementation (cool that you added the ability for tabs to wrap around!). And cool that you can drag tabs around!

But I'm afraid I wasn't clear. Actually what I am requesting is an improvement to default web browser behavior --I would like to cycle to the **previous** tab, rather than an adjacent tab. See:

1. https://superuser.com/questions/104917/chrome-tab-ordering/1139391#1139391
2. https://superuser.com/questions/290704/switching-back-to-last-tab-on-firefox

Might I suggest:

1. `Ctrl + Arrow` switch to adjacent tab
2. `Ctrl + Tab` switch to previous tab

I don't want to fit too many request in one issue, but I also have noticed these missing quality of life features:

1. No ability to drag a tab into a separate window
2. When dragging tabs, normally the tab moves with the mouse until dropped somewhere (providing some feedback); this does not occur in xschem.
3. The arrow keys are not present on the bindkeys cheat sheet

StefanSchippers, 2025-08-22:
I think ctrl-TAb now does what you ask. Pleas test.

Also a visual effect is added when dragging tabs. Please test.

The arrow keys are now added to the cheatsheet.

dragging a tab to a new separate window is now postponed. This need a huge work, and I don't know if it's worth doing.
Xschem currently support multiple windows or multiple tabs. Not both.

You can however start a new xschem instance and do whatever you want with it.

slagernate, 2025-08-22:
Awesome, thank you so much Stefan

slagernate, 2025-08-22:
What might also be cool is something like tree style tabs:

https://chromewebstore.google.com/detail/tree-style-tab/oicakdoenlelpjnkoljnaakdofplkgnd?hl=en

e.g. right now `e` will make you save the file, removing all history of changes (undo will no longer work). If `e` were to open the subcircuit in a new child tab, this history would not be removed (as both tabs would be open). (on another note, it may be nice to keep history of changes even after saving).