StefanSchippers/xschem #46: Question how to switch between schematic and PEX netlist

hpretl, 2022-01-10:
Hi Stefan,

What is the most efficient way for switching (in a test bench) between a schematic block (instantiated through the symbol), and a SPICE netlist of this circuit, generated by parasitic layout extraction? The interface (means pin list) is naturally the same in both cases.

I would like to build a single simulation test bench, which I can easily adapt/switch between both cases. I don't know if this is already available, but a flag or parameter attached to the symbol whether to netlist into the cell or include a spice file instead would probably do.

StefanSchippers, 2022-01-11:
Hi, 

### Solution 1
symbols may have a 'schematic' attribute that will cause xschem to descend into the specified schematic.

[image]

If i want an alternate description for the inverter i usually descend into the schematic, remove everything, keeping only the I/O pins, then saving to a different name, if the cell schematic is `test_inv.sch` i save it to `test_inv_pex.sch`.

[image]

Then i enter into the symbol (test_inv.sym) and by pressing 'q' in the drawing area i attach the following attribute:
`schematic="test_inv_pex.sch"`:

[image]

this will cause xschem to descend into `test_inv_pex.sch` instead of `test_inv.sch`.

The test_inv_pex.sch can be filled with a netlist by pressing 'q' in the drawing area and loading a parasitic extracted netlist.

[image]

If you extract the netlist you get the parasitic netlist code for the inverter, see the 'sch_path' comment line:
```
** sch_path: /home/schippes/x/test_open_pdks/test_pex.sch
**.subckt test_pex A Z1 Z2
*.ipin A
*.opin Z1
*.opin Z2
x8 A Z1 VCC VSS test_inv W_N=1 L_N=0.15 W_P=2 L_P=0.35 m=1
x1 A Z2 VCC VSS test_inv W_N=1 L_N=0.15 W_P=2 L_P=0.35 m=1
**.ends

* expanding   symbol:  test_inv.sym # of pins=2
** sym_path: /home/schippes/x/test_open_pdks/test_inv.sym
** sch_path: /home/schippes/x/test_open_pdks/test_inv_pex.sch
.subckt test_inv  a y  VCCPIN  VSSPIN      W_N=1 L_N=0.15 W_P=2 L_P=0.35
*.opin y
*.ipin a
**** begin user architecture code
**parasitic netlist follows
X0 Y A VCCPIN VCCPIN sky130_fd_pr__pfet_01v8_hvt w=840000u l=150000u
X1 Y A VSSPIN VSSPIN sky130_fd_pr__nfet_01v8 w=420000u l=150000u
X2 VCCPIN A Y VCCPIN sky130_fd_pr__pfet_01v8_hvt w=840000u l=150000u

**** end user architecture code
.ends
.end
```

If you remove the 'schematic attribute from the inverter symbol and redo the netlist you get the definition of `test_inv.sch`: 
```
** sch_path: /home/schippes/x/test_open_pdks/test_pex.sch
**.subckt test_pex A Z1 Z2
*.ipin A
*.opin Z1
*.opin Z2
x8 A Z1 VCC VSS test_inv W_N=1 L_N=0.15 W_P=2 L_P=0.35 m=1
x1 A Z2 VCC VSS test_inv W_N=1 L_N=0.15 W_P=2 L_P=0.35 m=1
**.ends

* expanding   symbol:  test_inv.sym # of pins=2
** sym_path: /home/schippes/x/test_open_pdks/test_inv.sym
** sch_path: /home/schippes/x/test_open_pdks/test_inv.sch
.subckt test_inv  a y  VCCPIN  VSSPIN      W_N=1 L_N=0.15 W_P=2 L_P=0.35
*.opin y
*.ipin a
XM2 y a VCCPIN VCCPIN sky130_fd_pr__pfet_01v8_lvt L=L_P W=W_P nf=1 ad='int((nf+1)/2) * W/nf * 0.29' as='int((nf+2)/2) * W/nf * 0.29'
+ pd='2*int((nf+1)/2) * (W/nf + 0.29)' ps='2*int((nf+2)/2) * (W/nf + 0.29)' nrd='0.29 / W' nrs='0.29 / W'
+ sa=0 sb=0 sd=0 mult=1 m=1
XM1 y a VSSPIN VSSPIN sky130_fd_pr__nfet_01v8_lvt L=L_N W=W_N nf=1 ad='int((nf+1)/2) * W/nf * 0.29' as='int((nf+2)/2) * W/nf * 0.29'
+ pd='2*int((nf+1)/2) * (W/nf + 0.29)' ps='2*int((nf+2)/2) * (W/nf + 0.29)' nrd='0.29 / W' nrs='0.29 / W'
+ sa=0 sb=0 sd=0 mult=1 m=1
.ends
.end
```

### Solution 2
you add a `subcircuit=...` attribute to your instances (you can do this to one, some, or all of them)
that specifies what subcircuit name to use:

[image]

Then you set this attribute (@subcircuit) instead of @symname in the symbol spice netlisting rule (the format attribute string):

[image]

The netlist will expand the test_inv.sch subcircuit, but the instances with the 'subcircuit' attribute set will reference another subcircuit.
The extracted netlist of all pex blocks can be included (for example using a 'code.sym' element in the toplevel.
Having the expansion of test_inv.sch even if it is not used (due to the subcircuit hijacking explained above) is usually not a problem for simulators.

```
** sch_path: /home/schippes/x/test_open_pdks/test_pex.sch
**.subckt test_pex A Z1 Z2
*.ipin A
*.opin Z1
*.opin Z2
x8 A Z1 VCC VSS test_inv_pex W_N=1 L_N=0.15 W_P=2 L_P=0.35 m=1
x1 A Z2 VCC VSS test_inv_pex W_N=1 L_N=0.15 W_P=2 L_P=0.35 m=1
**.ends

* expanding   symbol:  test_inv.sym # of pins=2
** sym_path: /home/schippes/x/test_open_pdks/test_inv.sym
** sch_path: /home/schippes/x/test_open_pdks/test_inv.sch
.subckt test_inv  a y  VCCPIN  VSSPIN  subcircuit  W_N  L_N  W_P  L_P  m     W_N=1 L_N=0.15 W_P=2
+ L_P=0.35
*.opin y
*.ipin a
XM2 y a VCCPIN VCCPIN sky130_fd_pr__pfet_01v8_lvt L=L_P W=W_P nf=1 ad='int((nf+1)/2) * W/nf * 0.29' as='int((nf+2)/2) * W/nf * 0.29'
+ pd='2*int((nf+1)/2) * (W/nf + 0.29)' ps='2*int((nf+2)/2) * (W/nf + 0.29)' nrd='0.29 / W' nrs='0.29 / W'
+ sa=0 sb=0 sd=0 mult=1 m=1
XM1 y a VSSPIN VSSPIN sky130_fd_pr__nfet_01v8_lvt L=L_N W=W_N nf=1 ad='int((nf+1)/2) * W/nf * 0.29' as='int((nf+2)/2) * W/nf * 0.29'
+ pd='2*int((nf+1)/2) * (W/nf + 0.29)' ps='2*int((nf+2)/2) * (W/nf + 0.29)' nrd='0.29 / W' nrs='0.29 / W'
+ sa=0 sb=0 sd=0 mult=1 m=1
.ends

.end

``` 

[test_open_pdks.tar.gz](https://github.com/StefanSchippers/xschem/files/7843006/test_open_pdks.tar.gz)

StefanSchippers, 2022-01-11:
### Solution 3

the `type` of a symbol is usually `subcircuit` if there  is a schematic to be traversed.
However setting a symbol type to `primitive` will cause the spice netlister to NOT descend into the schematic:

[image]

the netlist will just skip the inverter expansion:
```
** sch_path: /home/schippes/x/test_open_pdks/test_pex.sch
**.subckt test_pex A Z1 Z2
*.ipin A
*.opin Z1
*.opin Z2
x8 A Z1 VCC VSS test_inv W_N=1 L_N=0.15 W_P=2 L_P=0.35 m=1
x1 A Z2 VCC VSS test_inv W_N=1 L_N=0.15 W_P=2 L_P=0.35 m=1
**.ends
.end
```
All attributes in xschem can be passed to tcl by using the tcleval(...) construct:

`type=tcleval($env(sym_type))`

will cause the symbol to be treated as a subcircuit or a primitive depending on the `sym_type` environment variable.
`sym_type=primitive xschem test_pex.sch`
When the type is set to primitive you need to provide (by a .include usually) the extracted netlists to the simulator.

StefanSchippers, 2022-01-11:
Just another comment.
There is a special instance or symbol attribute called `device_model`: 

[image]

this attribute will be copied at the end of the netlist only once. If you have many components with a device_model attribute, it is copied in the netlist only once. 

```
** sch_path: /home/schippes/x/test_open_pdks/test_pex.sch
**.subckt test_pex A Z1 Z2
*.ipin A
*.opin Z1
*.opin Z2
x8 A Z1 VCC VSS test_inv W_N=1 L_N=0.15 W_P=2 L_P=0.35 m=1
x1 A Z2 VCC VSS test_inv W_N=1 L_N=0.15 W_P=2 L_P=0.35 m=1
**.ends
**** begin user architecture code
.include /home/schippes/share/pdk/blablabla
**** end user architecture code
.end
```

if you have 10 inverters with `device_model=".include inv.spice"` and 10 nand gates with `device_model=".include nand.spice"`  the netlist will contain one .include per type.

StefanSchippers, 2022-01-11:
Harald, i tried this example:

[image]

the code element dumps the content of shell variable `include_pex`, and the symbol inverter has the type attribute set to `tcleval($env(sym_type))`; the result is that if i call xschem in the following way: 

`include_pex='.include pex_netlist.spice' sym_type=primitive  xschem test_pex.sch`

The subcircuits are not expanded and a `.include pex_netlist.spice` is inserted in the netlist.

If i call xschem the following way:

`include_pex='* .include pex_netlist.spice' sym_type=subcircuit  xschem test_pex.sch`

The original xschem netlist is generated, expanding the inverter sub-schematics.

hpretl, 2022-01-12:
Thanks for all the hints! I experimented a bit, and my favorite workflow is now:

* I copy the symbol and schematic of a given cell (e.g. `inv.sym` and `inv.sch`) to e.g. `inv.pex.sym` and `inv.pex.sch`.
* I empty the schematic of `inv.pex.sch`, just leave pins and the `NC` terminations in.
* I put in a `code.sym`, where I put in the `inv.pex.spice`.

Then I just need to change the instantiated symbol to switch between schematics and PEX netlist. The whole flow is also simple enough that it can be automated in a script (or by `xschem` functionality :-).

[image]

[image]

StefanSchippers, 2022-01-21:
closing for now, feel free to reopen if needing more functionality.

closed 2022-01-21 by StefanSchippers