StefanSchippers/xschem #47: Enabling "Split netlist" causes Xschem to crash on netlist generation

Unkraut3, 2022-01-17:
Hello,

I'm using the option "Split netlist" (menu "Options", then tick "Split netlist") when generating ngspice netlists. I use this feature to work around a limitation of ngspice. 

On an irregular basis Xschem crashes when pressing the "Netlist" button if "Split netlist" is enabled. The behaviour is difficult to reproduce. Sometimes Xschem crashes immediately. Sometimes, I can export several netlists without any crashes. I have attached a schematic with which I can observe the bug: [split_netlist_bug.zip](https://github.com/StefanSchippers/xschem/files/7880464/split_netlist_bug.zip).

Versions of Xschem where the issue appears: 3.0.0 and the current HEAD.

StefanSchippers, 2022-01-17:
Thank you, will look at this immediately!

StefanSchippers, 2022-01-17:
I  found the issue, (it was a bit difficult to find out), there was some writing to netlist file where the netlist file was already closed, this happened only in split netlist mode, indeed.
After the fix i am no more able to crash xschem. Could you please make some tests and report back if OK or still failing ? thanks

Unkraut3, 2022-01-19:
Thank you! Now the netlist generation works flawless. I used it now for some hours without any crash. 

Grazie mille per questo software eccezionale!

StefanSchippers, 2022-01-21:
Fixed.

closed 2022-01-21 by StefanSchippers