StefanSchippers/xschem #48: Problem when configuring

zhengshuaige66, 2022-02-28:
Hi Stefan,
       When configure xschem, an error code appeared “Node libs/script/tk/* is required but provided detection callback fails to find that feature on that system.” But I have installed libtcl8.6(8.6.11+dfsg-1)，libtk8.6(8.6.11-2build1) and tk8.6/tcl8.6(8.6.11+1build1) .
       Thank you.

[image]

StefanSchippers, 2022-02-28:
you probably need to install tk8.6-dev and tcl8.6-dev packages:
sudo apt-get install tcl8.6-dev tk8.6-dev. Then run configure again.
Also see complete list of required libraries [here](https://xschem.sourceforge.io/stefan/xschem_man/install_xschem.html).
Let me know if something else does not work.

zhengshuaige66, 2022-03-01:
I checked the library and it looks like the required library is indeed installed.

> [image]

> [image]

I observed the information in other people's issues and found the following differences.

> [image]

> [image]

This looks like it found tk8.6 but can't locate it

zhengshuaige66, 2022-03-01:
Stefan,
     Thank u very much. I downloaded the latest installation package from the tk/tcl official website for installation(both tk and tcl need to be installed), and now I can install xschem normally.

closed 2022-03-01 by zhengshuaige66