StefanSchippers/xschem #49: ngspice simulator not opens

visionvlsi, 2022-03-01:
I have followed the latest video tutorial. And followed the old video tutorial for ngspice. After launching the tool by typing xschem, after clicking on the netlist then i clicked simulate, but nothing happens. whereas .spice file been created in simulations folder.

visionvlsi, 2022-03-01:
The following is the nmos inverter netlist 

** sch_path: /usr/local/share/pdk/sky130A/libs.tech/xschem/sky130_tests/test_nmos.sch
**.subckt test_nmos G1v8 D1v8 B
*.ipin G1v8
*.ipin D1v8
*.ipin B
Vd1 D1v8 net1 0
.save  i(vd1)
Vd3 D3v3 net2 0
.save  i(vd3)
Vd2 D1v8 net3 0
.save  i(vd2)
Vd4 D5v0 net4 0
.save  i(vd4)
E1 D5v0 0 D1v8 0 '5/1.8'
E2 D3v3 0 D1v8 0 '3.3/1.8'
E3 G5v0 0 G1v8 0 '5/1.8'
E4 G3v3 0 G1v8 0 '3.3/1.8'
Vd5 D10v5 net5 0
.save  i(vd5)
E5 D10v5 0 D1v8 0 '10.5/1.8'
E6 D16v0 0 D1v8 0 '16.0/1.8'
Vd6 D16v0 net6 0
.save  i(vd6)
Vd7 D1v8 net7 0
.save  i(vd7)
E7 D20v0 0 D1v8 0 '20.0/1.8'
XM1 net1 G1v8 S B sky130_fd_pr__nfet_01v8_lvt L=0.15 W=1 nf=1 ad='int((nf+1)/2) * W/nf * 0.29' as='int((nf+2)/2) * W/nf * 0.29'
+ pd='2*int((nf+1)/2) * (W/nf + 0.29)' ps='2*int((nf+2)/2) * (W/nf + 0.29)' nrd='0.29 / W' nrs='0.29 / W'
+ sa=0 sb=0 sd=0 mult=1 m=1
XM2 net3 G1v8 S B sky130_fd_pr__nfet_01v8 L=0.15 W=1 nf=1 ad='int((nf+1)/2) * W/nf * 0.29' as='int((nf+2)/2) * W/nf * 0.29'
+ pd='2*int((nf+1)/2) * (W/nf + 0.29)' ps='2*int((nf+2)/2) * (W/nf + 0.29)' nrd='0.29 / W' nrs='0.29 / W'
+ sa=0 sb=0 sd=0 mult=1 m=1
XM3 net2 G3v3 S B sky130_fd_pr__nfet_03v3_nvt L=0.5 W=1 nf=1 ad='int((nf+1)/2) * W/nf * 0.29' as='int((nf+2)/2) * W/nf * 0.29'
+ pd='2*int((nf+1)/2) * (W/nf + 0.29)' ps='2*int((nf+2)/2) * (W/nf + 0.29)' nrd='0.29 / W' nrs='0.29 / W'
+ sa=0 sb=0 sd=0 mult=1 m=1
XM4 net4 G5v0 S B sky130_fd_pr__nfet_05v0_nvt L=0.9 W=1 nf=1 ad='int((nf+1)/2) * W/nf * 0.29' as='int((nf+2)/2) * W/nf * 0.29'
+ pd='2*int((nf+1)/2) * (W/nf + 0.29)' ps='2*int((nf+2)/2) * (W/nf + 0.29)' nrd='0.29 / W' nrs='0.29 / W'
+ sa=0 sb=0 sd=0 mult=1 m=1
XM5 net5 G5v0 S B sky130_fd_pr__nfet_g5v0d10v5 L=0.5 W=1 nf=1 ad='int((nf+1)/2) * W/nf * 0.29' as='int((nf+2)/2) * W/nf * 0.29'
+ pd='2*int((nf+1)/2) * (W/nf + 0.29)' ps='2*int((nf+2)/2) * (W/nf + 0.29)' nrd='0.29 / W' nrs='0.29 / W'
+ sa=0 sb=0 sd=0 mult=1 m=1
XM6 net6 G5v0 S B sky130_fd_pr__nfet_g5v0d16v0 L=0.7 W=5.0 nf=1 ad='int((nf+1)/2) * W/nf * 0.29' as='int((nf+2)/2) * W/nf * 0.29'
+ pd='2*int((nf+1)/2) * (W/nf + 0.29)' ps='2*int((nf+2)/2) * (W/nf + 0.29)' nrd='0.29 / W' nrs='0.29 / W'
+ sa=0 sb=0 sd=0 mult=1 m=1
Vd8 D1v8 net8 0
.save  i(vd8)
XM7 net7 G1v8 S B sky130_fd_pr__nfet_20v0 L=2.95 W=29.41 m=1
XM16 net8 G1v8 S B sky130_fd_pr__nfet_20v0_zvt W=30 L=5 m=1
**** begin user architecture code

* this option enables mos model bin
* selection based on W/NF instead of W
.option wnflag=1
*.option savecurrents
vg G1v8 0 1.8
vs s 0 0
vd D1v8 0 1.8
vb b 0 0
.control
save @m.xm1.msky130_fd_pr__nfet_01v8_lvt[gm]
save all
dc vd 0 1.8 0.0005 vg 0 1.8 0.2
write test_nmos.raw
* dc vd 0 1.8 0.001 vg 0 1.2 0.1
plot all.vd1#branch vs D1v8
*plot all.vd2#branch vs D1v8
plot all.vd3#branch vs D3v3
*plot all.vd4#branch vs D5v0
plot all.vd5#branch vs D10v5
*plot all.vd6#branch vs D16v0
plot all.vd8#branch vs D1v8
plot all.vd7#branch vs D1v8
set appendwrite
op
write test_nmos.raw
.endc

** opencircuitdesign pdks install
.lib /usr/local/share/pdk/sky130A/libs.tech/ngspice/sky130.lib.spice tt

**** end user architecture code
**.ends
.end

StefanSchippers, 2022-03-01:
First of all open a terminal and type 'ngspice'. Does the simulator start?

visionvlsi, 2022-03-01:
Thank you I solved that problem by installing 

sudo apt-get install -y xterm

closed 2022-03-01 by visionvlsi

StefanSchippers, 2022-03-01:
Well done!!