StefanSchippers/xschem #51: Problems when using hotkeys

zhengshuaige66, 2022-03-14:
Hi Stefan,
&emsp; I found some problems when I used the hotkey to draw the schematic. When I use 'w' to insert the wire , the beginning of the wire always starts from the top left corner, and I can't draw the wire properly. But when I insert the wire using the right-click menu, it's back to normal.The dynamic diagram of the line drawing I put below.
[image]
&emsp; A similar thing happens when I use a hotkey to move a device, after clicking 'm', the device position changes to the bottom right corner, I need to move the mouse to the upper left corner to see it. It is also normal to use the right-click menu to move the device.
[image]
&emsp; Please how can I solve this problem, thank you！

StefanSchippers, 2022-03-14:
hello, thank you for reporting this issue. what system are you using? I am looking back in the logs, one user had similar isues some years ago, need to recall this...

zhengshuaige66, 2022-03-15:
Hello, the following are the various dependency package versions of xschem that I have installed.
[image]
And the following are the system distribution and system kernel version I have installed.
[image]
[image]

StefanSchippers, 2022-03-15:
This is probably an issue with the Xserver. Are you using Wayland + Xwayland compatibility layer or are you using your desktop natively with the Xorg server?

StefanSchippers, 2022-03-15:
I have a test for you:
- Open a terminal
- type `wish` (or `wish8.6`, depending on the linux system)
- at the wish prompt type this command:  `bind . <KeyPress> { puts "%x %y"}`
- now position the mouse inside the wish empty light-grey window and press a key on the keyboard.
- Does wish print the x,y coordinates of the mouse? or does it print `0 0`?

zhengshuaige66, 2022-03-15:
The default Xserver of my system is X11, and when I switch it to Wayland, the problem still exists.
I did what you did and I got the following display.
[image]

StefanSchippers, 2022-03-15:
Thank you very much for doing this test. 
You see the problem. The little test should print the x,y coordinates of the mouse inside the white wish window, relative to the origin in the upper-left corner of the window, y axis going downward and x going from left to right. 
Mouse coordinates should always be positive if the mouse pointer is inside the white area. 
The negative coordinates you are seeing confuse xschem. When you press 'w' to draw a wire the x,y coordinates are taken as the origin of the wire. Since on your system you get negative values the program gets confused. 
I need to investigate upstream, this is not a problem of xschem by itself but it's a problem in the tcl-tk libraries or in the xserver.
Will keep you informed if i get some results.
Thank you for your help

StefanSchippers, 2022-03-15:
Just another info: do you know what graphic card  (GPU) do you have?
you can use `lspci` command, on my system for example i get this (AMD radeon):
```
schippes@mazinga:~/xschem-repo/trunk/src$ lspci
...
...
01:00.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Park [Mobility Radeon HD 5430/5450/5470]
...
...

```
If you don't have this `lspci` command just install package `pciutils`, it is very small and useful.
Thanks

StefanSchippers, 2022-03-29:
I have checked bug history and the other user having this very same issue was also using vmware. That user then tried the same setup on a native ubuntu system and the problem was gone. SO it appears vmware Xserver is reporting wrong X,Y coordinates on keypress events. If i find some solution in the VMware support i will let you know.
If you have time to experiment you can also check Virtualbox. I used VBox and works fine.

zhengshuaige66, 2022-04-03:
Stefan,
    Thank you a lot. The problem does appear on Vmware, I upgraded Vmware to the latest version and reinstalled Ubuntu and xschem, and it can be used normally now.

closed 2022-04-03 by zhengshuaige66