StefanSchippers/xschem #52: XSchem Command mode for schematic entry

AhmedLilah, 2022-04-03:
I'm looking to use xschem in command mode only but there doesn't seem to be a comprehensive doc for the commands .

if there is a doc for how to use command mode for schematic entry among all the other uses please refer to it. otherwise, your help will be very appreciated if you can make such a document.

StefanSchippers, 2022-04-05:
Thank you, understood.
You can also directly create a .sch file, it is a plain ascii file. You may create in xschem a simple schematic with one component and some wires, save it and open the generated .sch file with an editor.
The file format is documented in the online [manual](https://xschem.sourceforge.io/stefan/xschem_man/developer_info.html)

AhmedLilah, 2022-04-07:
that's great.
is the ASCII syntax for the .sch ad .sym files fully documented cause that may remove the need for the remote commands entirely for me

StefanSchippers, 2022-04-08:
Currently there is no option for user specified width for lines. 
There is an exception: for wires and lines a `bus=true` attribute draws the lines/wires at 3x width. This is used to draw multi-bit buses. 
Regarding the question about file format specification i believe it is fully documented in the manual ( https://xschem.sourceforge.io/stefan/xschem_man/developer_info.html) . If you find something unclear please report i will make the necessary corrections.

AhmedLilah, 2022-04-09:
thanks, the bus hack proved to be of good use, I'll try to give all the feedback on anything that I encounter.

also is there a way to set the default config to enable stretch, set theme to bright and layer color, etc for any new schematics or symbols?

StefanSchippers, 2022-04-09:
yes, you can set the desired config, add these 2 lines in your xschemrc (typically in `~/.xschem/xschemrc`):
```
set enable_stretch 1
set dark_colorscheme 0
```

AhmedLilah, 2022-04-09:
is there like a list for the remaining xschemrc cofig parameters? 
ex: enable_remember_last_command 
etc..

also can i add comments in the .sch and .sym files?  couldn't find info on that in the developers info 
if yes how?

Does xschem support cadence schematics?

StefanSchippers, 2022-04-25:
For comments you can add a text object. You can set it to hidden if you add an attribute hide=true in the props textbox when editing and setting up the text object with the dialog box.

Another method is to add manually (using a text editor) a comment in the version string, see below:

change this line: 
```
v {xschem version=3.0.0 file_version=1.2 }
```
to: 
```
v {xschem version=3.0.0 file_version=1.2 
* Copyright 2021 Stefan Frederik Schippers
* 
* Licensed under the Apache License, Version 2.0 (the "License");
* you may not use this file except in compliance with the License.
* You may obtain a copy of the License at
*
*     https://www.apache.org/licenses/LICENSE-2.0
*
* Unless required by applicable law or agreed to in writing, software
* distributed under the License is distributed on an "AS IS" BASIS,
* WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
* See the License for the specific language governing permissions and
* limitations under the License.
}
```
I will add this in the developer info page.

closed 2022-06-22 by StefanSchippers