StefanSchippers/xschem #53: bus=true does not render in symbol lines

AhmedLilah, 2022-04-09:
while making my own symbols I used the bus=true on some of the lines it works fine on windows
[image]

but in Linux (wsl2 Debian) the line is not rendered at the 3x width in either the symbols file or in the schematic file 
[image]

StefanSchippers, 2022-04-09:
can you please open a new schematic in linux, draw 2 lines and set bus=true
attribute on one of these, do they look different?

AhmedLilah, 2022-04-09:
> aking my own symbols I used the bus=true on som

the top line have is a bus

[image]

AhmedLilah, 2022-04-09:
also the line width does not work on windows and in both OS the lines do get a lot thinner without any mods while zoning and redraw doesn't return them back have to close the program and open it again

StefanSchippers, 2022-04-10:
Did you install xschem from source files ? because packaged xschem versions in ubuntu/debian are too old. 
If i draw 2 lines and set bus=true on one of them this is what i get:
[image]
Ensure menu option View->Toggle variable line width is selected.
If you zoom out a lot of course the lines will get 1 pixel thin and you won't see any difference. This is intended behavior.

closed 2022-06-22 by StefanSchippers