StefanSchippers/xschem #54: Schematic files use absolute paths

tdene, 2022-04-10:
# Reproduction steps

- Create a schematic file
- Insert a user-defined symbol (previously created from another xschem schematic file)
- Save the schematic
- Close xschem
- Open the schematic in a plain-text editor
- Note that the path to the user-defined symbol is an absolute path

# Why this is a problem

Sharing schematics by moving them to a different folder or uploading them to GitHub becomes difficult due to the need to edit the absolute paths in plain-text

# Suggested fix

Introduce a flag allowing for a choice between absolute and relative paths. Or introduce a script that will convert all absolute paths to relative ones in preparation for public upload.

StefanSchippers, 2022-04-10:
Absolute paths are used if you insert a symbol from an unknown directory, that is, a directory that is not part of the XSCHEM_LIBRARY_PATH defined in xschemrc.

If you want to make the 'current directory' part of the search paths just include '.' in the list of paths in XSCHEM_LIBRARY_PATH defined in xschemrc:
Please remember that '.' in xschem means 'the directory of the parent schematic'
```
set XSCHEM_LIBRARY_PATH {}
append XSCHEM_LIBRARY_PATH :.
append XSCHEM_LIBRARY_PATH :/some/path
append XSCHEM_LIBRARY_PATH :/other/path
```
With this setup no absolute paths are saved.

tdene, 2022-04-11:
@StefanSchippers thank you greatly for the tips! My group is just starting to use xschem and still learning it!

StefanSchippers, 2022-04-16:
Closing for now. If you still experience issues feel free to submit and/or reopen.

closed 2022-04-16 by StefanSchippers