StefanSchippers/xschem #55: problem in simulation

ripudamank2, 2022-04-24:
Hi, 
I had  installed latest version of xschem and sky130a by watching video. 
When i tried to run one basic circuit from the example it was given below error which is attached .
Can you please tell the solution to this problem?
[image]

StefanSchippers, 2022-04-24:
Hi, did you watch [this video](https://www.youtube.com/watch?v=KgBLByOkJxA)? The standard way of using sky130 with xschem now is using [open_pdks](http://opencircuitdesign.com/open_pdks/index.html). Installing open_pdks will take care of downloading all spice models, making various corrections and also install xschem_sky130 libraries.
Do not directly install skywater-pdk, let open_pdks do this for you.

ripudamank2, 2022-04-25:
yes its working fine , thank you so much