StefanSchippers/xschem #56: Paremeter handling in Verilog I/O declaration

rbarzic, 2022-04-25:
Hi,

I'm making a schematic where I would like some I/Os to have a width that depends from a parameter.
I give the pin name a lab parameter like:
`name=p1 lab=sel_delay[RIPPLE_COUNTER_BITS-1:0]`
When I generate the netlist, I get the pin generated correctly:

```verilog
module vcotempsensor_dig_delays (
  output wire rdy_osc1_delay,
  output wire rdy_osc2_delay,
  output wire rdy_n_az_delay,
  output wire rdy_n_startup_delay,
  output wire rdy_n_acc2_delay,
  input wire en_osc1_delay,
  input wire en_osc2_delay,
  input wire en_n_az_delay,
  input wire en_n_startup_delay,
  input wire en_n_acc2_delay,
  input wire rst_n,
  input wire clk_32khz,
  input wire [RIPPLE_COUNTER_BITS-1:0] sel_delay
);
parameter LIB = "UNDEFINED" ;
parameter RIPPLE_COUNTER_BITS = 8 ;
wire net6  ;
wire net12  ;

```

I used the following  attributes in the matching symbols:

```
type=async_lib
format="@name @pinlist @symname"
template="name=x1 LIB=UNDEFINED RIPPLE_COUNTER_BITS=8"
generic_type="LIB=string"
```
How can I get the verilog parameters to be defined before the input/output list (verilog-2001 style):
```verilog
module vcotempsensor_dig_delays 
#(
parameter LIB = "UNDEFINED",
parameter RIPPLE_COUNTER_BITS = 8 
)
(
 output wire rdy_osc1_delay,
  output wire rdy_osc2_delay,
  output wire rdy_n_az_delay,
....
 input wire [RIPPLE_COUNTER_BITS-1:0] sel_delay
)
```
I tried to play with the verilog_format attribute but it does seem to have an effect there

StefanSchippers, 2022-04-25:
Hi, Ronan, 

I have a similar example (it's in std xschem distribution examples,` logic/testbench.sch`). I have the exact parametrized bus width on a port, and parameters specified in the 1995 veriilog format. However simulator (icarus verilog) runs fine. 
Are some tools refusing this pre-2001 syntax for parameters? 

Anyway it's a good point. If verilog2001 option is enabled in the 'Options' menu i will switch the parameter declaration as you suggested.
Thanks

StefanSchippers, 2022-04-25:
Please update your xschem, can you please test? I have changed the module parameter declaration as you suggested if verilog2001 is enabled in Options menu (the default setting).
```
// expanding   symbol:  ram.sym # of pins=8
// sym_path: /home/schippes/xschem-repo/trunk/xschem_library/logic/ram.sym
// sch_path: /home/schippes/xschem-repo/trunk/xschem_library/logic/ram.sch
`timescale 1ps / 1ps
module ram
#(
  parameter dim = 5,
  parameter width = 8,
  parameter hex = 0,
  parameter datafile = "ram.list",
  parameter modulename = "ram",
  parameter access_delay = 3000,
  parameter oe_delay = 300
)
(
  input wire [dim-1:0] ADD,
  input wire [width-1:0] M,
  output wire [width-1:0] DOUT,
  input wire [width-1:0] DIN,
  input wire WEN,
  input wire CEN,
  input wire OEN,
  input wire CK
);
```

closed 2022-06-22 by StefanSchippers