StefanSchippers/xschem #57: XOpenDisplay... not found

Talha-Ahmed-1, 2022-06-15:
I am configuring xschem it shoes me this error after running the following command.
```
./configure --prefix=/usr/local --user-conf-dir=~/.xschem --user-lib-path=~/share/xschem/xschem_library --sys-lib-path=/usr/local/share/xschem/xschem_library
```
```
make: Nothing to be done for 'all'.
Setting prefix to '/usr/local'
--- Detecting host
Looking for a shell... /bin/sh -c
Looking for a shell escape character... found: '\'
Detecting temp dir... validated /home/talha/xschem/scconfig/
Checking for system type... OK (name: Linux; class: UNIX)
--- Detecting target (same as host)
Checking for cc (target)... OK ('gcc', ' -pipe', '')
Checking for -fpic... OK (-fPIC)
Checking for popen(3)... OK ('#define _XOPEN_SOURCE\n', '' and '')
Checking for dup2(2)... OK ('', '' and '')
Checking for flex... Found.
Checking for bison... Found.
Checking for XOpenDisplay... not found
Node libs/gui/xopendisplay/cflags is required but provided detection callback fails to find that feature on that system.
Aborted (core dumped)

```

StefanSchippers, 2022-06-22:
You need to install the required libraries as explained here:
https://xschem.sourceforge.io/stefan/xschem_man/install_xschem.html
After installing missing stuff run ./configure again.

closed 2022-06-22 by StefanSchippers

StefanSchippers, 2022-10-11:
you need to install all required libraries first. Check the list here:
https://xschem.sourceforge.io/stefan/xschem_man/install_xschem.html