StefanSchippers/xschem #62: Exporting PDF with waveforms in schematic

rafmag, 2022-07-30:
I found an issue when trying to export a PDF of a schematic that contains an embedded waveform graph. The graph does not get exported.

StefanSchippers, 2022-07-31:
Yes this needs to be implemented. It's on the list.
Embedded graphs may contain lot of data (for example multiple traces of a 50000 points transient analysis). We will see how efficiently pdf handles such vector data.

rafmag, 2022-07-31:
I see the issue. I wonder if there is a way to rasterize just the graphs?

StefanSchippers, 2022-07-31:
This is of course one of the options.

StefanSchippers, 2022-09-15:
SVG export images now contain graphs (rendered as embedded png images. Since the code is in place for svg i think the road is paved for doing the same for ps/pdf. I need to implement the ASCII85 encoding/decoding that is used in postscript instead of the more common base64.
[image]

rafmag, 2023-01-13:
This is awesome. Has this change been committed? Its not working for me.

StefanSchippers, 2023-01-13:
for svg exports its working, if you have cairo libs installed.
is svg export not working for graphs on your system?
ensure you have   libcairo2 and libcairo2-dev packages installed while building xschem.

rafmag, 2023-01-13:
my mistake, I was running it on a version of xschem that wasn't updated. It's working for SVG.

StefanSchippers, 2023-01-13:
in my spare time i am reading the adobe postscript books and lurking into some .ps files. One day iI will be able to embed images/graphs in postscript/PDF as well. Postscript seems to be engineered to make things super complicated and difficult.

rafmag, 2023-01-13:
That would be nice. Right now it's not holding anything up for me, but it would be really nice to generate hierarchical PDFs with embedded waveforms and PNGs to document an IC design automatically in a single PDF.

StefanSchippers, 2023-01-16:
Thank you I have merged the changes, will work on the build system and libjpeg detection. Thanks!!

rafmag, 2023-01-16:
Just submitted an issue, didn't know you were working on it :)

StefanSchippers, 2023-01-16:
I am running some tests. So far only two issues are present. 
first one:
Images inside rotates symbols are not transformed (rotated/flipped).

schematic:
[image]
ps/pdf output:
[image]

test schematic:
[test_images2.zip](https://github.com/StefanSchippers/xschem/files/10425996/test_images2.zip)

Second issue is related to images xschem has read from jpg files and transformed to png by using a flter attribute. 
i am looking into this one.

rafmag, 2023-01-16:
I can probably tackle the first issue. Not so sure about the second.

StefanSchippers, 2023-01-16:
sorry second one is not existing, was my fault.

rafmag, 2023-01-16:
It was my first time trying something like this. It was a fun challenge!

StefanSchippers, 2023-01-16:
Can you please generate a ps/pdf export from this schematic file? 
on my side it generates invalid postscript code. If I remove one of the two images export goes Ok. Looking into it, now taking a break...
[test_images3.zip](https://github.com/StefanSchippers/xschem/files/10426169/test_images3.zip)

StefanSchippers, 2023-01-16:
For debugging if you specify a .ps output file xschem will not try to do the ps->pdf conversion...

rafmag, 2023-01-16:
`Can you please generate a ps/pdf export from this schematic file?
on my side it generates invalid postscript code. If I remove one of the two images export goes Ok. Looking into it, now taking a break...
`
The problem is with the larger image. it doesn't even display on my xschem.  When I trace the code, I'm getting an error on the filter_data code. I didn't quite understand what this filter stuff was all about.

StefanSchippers, 2023-01-16:
Thanks for checking. The filter code is used to pipe alien image formats and convert to png.
since cairo handles png images (and with your additions also jpeg, but before that i only had png) i looked into a way to read a jpg file:
by specifying:

filter="gm convert - png:-"
ximage=/home/schippes/x/1.jpg 

into a rectangle box the specified jpg image is read but before passing the data to cairo for rendering / svg exporting it is piped to the filter program, that is a gm process that convert any stdin data to png on stdout.

rafmag, 2023-01-16:
ah I was missing graphicsmagick I can now replicate the issue... but its very mysterious. 
I fixed the rotation, flip issue. There was also a small bug in draw.c when drawing scaled pngs that where rotated, but it's fixed now.

Let me figure out this filter bug now.

StefanSchippers, 2023-01-16:
yes if you look in my last committed psprint.c there is a #define FFLUSH_PS that adds some fflush() in the ps prints.  With this macro enabled i get correct postscript output. However this is not the solution, it means indeed there is some buffer overflow somewhere... or some buffer being used after freeing it.... by experience these are the things that usually happen...

rafmag, 2023-01-16:
Found it!!
There was a buffer issue when writing the very big HEX encoded strings into the PS file.
I fixed it by adding `fflush(fd);` at the end of the function.

Now I have to learn how to get github to ignore all the build files so I can push a clean version.

StefanSchippers, 2023-01-16:
yes, i got the same, however it is perfectly ok to free a large buffer
immediately after ffprintf()-ing  it out, so i  i am suspcious.

rafmag, 2023-01-16:
Alight I committed the fixes. I don't like the way low resolution PNGs are converted to jpgs. Scaled PNGs look much better.
[TEST_SCHEM.zip](https://github.com/StefanSchippers/xschem/files/10428689/TEST_SCHEM.zip)

[image]

StefanSchippers, 2023-01-16:
For removing untracked files in your local repository do:
`git rm Makefile.conf config.h doc/manpages/xschem.1 scconfig/config.cache scconfig/config.log scconfig/configure scconfig/sccbox src/Makefile`
then:
`git commit -a -m 'removed untracked files'`
then 
`git push`

rafmag, 2023-01-16:
ok, I think it's good for next time. I also added them to .gitignore, Well see.

StefanSchippers, 2023-01-16:
Thank you! rotation also works perfectly.
The only one remaining thing is the fflush() issue. These should not be necessary, will see if there is something more serious.

rafmag, 2023-01-16:
I mean there is not much more to it. I stepped through the code and tried to fclose the file right after fprinting the hex string. if I did it without the fflush the string would be incomplete, if I added the fflush then the string would be complete. Its like fprint/fwrite has some limitation with super long strings.

rafmag, 2023-01-16:
BTW, if we can implement Adobe ASCII85 encoding it would be more efficient (shorter string) than hex encoding and still compatible with the code. We would just have to change the `/ASCIIHexDecode` keyword. I was just too lazy to figure out the algorithm.

StefanSchippers, 2023-01-16:
If i generate a postscript file and look into it i have  this portion:
```
0.664062 0.132812 0.132812 RGB
0.664062 0.664062 0.664062 RGB
94 0.996094 RGB
0 0.265625 0.929688 RGB
```
the 3rd line is an impossible situation. This can only be explained by corruption, as the function that writes the RGB (setrgb) lines prints 3 doubles and RGB:

```
static void set_ps_colors(unsigned int pixel)
{
   if(color_ps) fprintf(fd, "%g %g %g RGB\n",
     (double)ps_colors[pixel].red/256.0, (double)ps_colors[pixel].green/256.0,
     (double)ps_colors[pixel].blue/256.0);    
}   
```

rafmag, 2023-01-16:
Do you have an example schematic. I can't replicate that issue with my schematics.

StefanSchippers, 2023-01-16:
Yes the corruption is random.Of course this happens if no fflush is present in the psprint.c file. I am investigating since *all* computer gurus have told me with 100.00000% confidence that any string buffer can be deleted right after fprintf, without any fflush. Even if I fix theRGB lines there are other corrupted sections, so the exact layout of the error is pointless.
I have had many times where a change in one file leads to failure in some other totally uncorrelated part of the program. These are the worst things that can happen, since the error tells nothing about the origin. I am almost sure the filter_data() function is causing the problem. If I can't get point out the problem i will crush this function away and remove the ability to read in files that are not png, and may be if we want to read jpg, use the cairo_jpg.c functions you have provided.

rafmag, 2023-01-16:
But fflush fixes it? what is the down side of fflush?

StefanSchippers, 2023-01-16:
fflush fixes it, but my fear is it fixes it just by 'miracle'. the flush changes the timings and may be we are lucky. However every software engineer i have asked is telling that it is *impossible* that fflush fixes such a problem, and if it does is just coincidence.

StefanSchippers, 2023-01-16:
I can not believe that Stefan Schippers found a bug in libc, such that fprintf does not buffer in the data before the data is subsequently freed. fprintf must block until all provided data is piped into the stdout stream.

rafmag, 2023-01-17:
It's definitely a buffer issue. If you disable the buffer with `setbuf(fd, NULL);`  after opening the file but before any fprints, there are no errors.

Clean enough solution?

StefanSchippers, 2023-01-17:
i have created another example with two images in a schematic, one extremely big, one smaller. The ps export fails if i keep both images. If I keep only one (even the super big one) export works fine. Will try to understand what causes the issue.

StefanSchippers, 2023-01-17:
Hi, Raf, 
In latest commit i have fixed the `filter_data()` function.
this function takes an arbitrary char buffer `din[]` of length `ilen` and a unix filter command `cmd` (something that takes data on stdin and spits out transformed data on stdout), and refurns the transformed char buffer `dout[]` of length `olen`.
This filter_data function creates a child process runs the cmd command into it, sends din data to it and reads dout data from it.

Previously the child `cmd` was run using the `system()` command. However this is implemented by running a subshell (/bin/sh) and giving the shell (via the -c argument) the cmd to execute. This adds one layer of uncertainty. If the shell by itself prints something on stdout it will break the whole protocol adding / mixing data with `cmd`'s stdout. So I have removed the `system()` call and used the lower level `execvp()` function. For this to work the cmd string must be split into individual arguments but this is easy.
You may look at `save.c --> filter_data()` for details.
After this change no `fflush()` calls are anymore needed, i have removed them in `psprint.c`.

Again THANK YOU for this great addition, I am so happy to have pdf working with graphs and images.
A hierarchical design can be exported altogether in a multipage pdf and symbols can be clicked bringing you to the corresponding schematic page. Super cool.

StefanSchippers, 2023-01-17:
A side note: You have defined a very useful InvertOnExport attribute to invert image colors. This is very useful for ps export since it uses the light colorscheme (white background). I have changed its allowed values to true/false instead of 1/0. This is just to keep a consistent format with all attributes.
Attributes that can only take a boolean value are set as `attr=true` or `attr=false`.
Attributes that take a number use `attr=1, attr=0, attr=3.141, ....` So if you have some definitions change 
InvertOnExport=1to InvertOnExport=true.

StefanSchippers, 2023-01-18:
I have also added a `jpeg_quality` attribute that specifies the quality of embedded jpg in postscript output. If unspecified (default) 100 (maximum quality, maximum file size) is used.
Using a value of 25 will reduce image size by 5/6 times and in most cases acceptable results so it is worth using.

StefanSchippers, 2023-01-18:
> the system() issue seems very platform specific, would this still work in
> a windows build? should you still include the fflush as a safety precaution?

I think i will put a setbuf instruction in ps write code to disable
buffering, as you also wrote it does the job.

StefanSchippers, 2023-01-19:
Status update: 

The corruption in postscript output was due to the filter_data() function. This function forks a child process that translates some arbitrary image formats to png. The child incorrectly called exit() when done.

This is wrong and child processes should call _exit() when finished. (http://www.unixguide.net/unix/programming/1.1.3.shtml)

The reason is that exit() flushes all pending stdio buffers, and since filter_data() is called while writing the postscript file there is certainly some unflushed data.
Note that child processes inherit from parent all open file descriptors. If the child exits with exit() it will flush pending data in the postscript output file. When the parent adds lines to the postscript file the pending data is flushed another time. This matches exactly what i saw in the corrupted postscript file: some random sections were printed twice.
the _exit() function does not flush pending stdio buffers.
With all these investigations i have done two things:
- fflush all stdio buffers before calling any fork() system call, so even if a child exits with exit() no data needs to be flushed in the parent.
- exit childs with _exit()

rafmag, 2023-01-19:
That's a crazy bug, I don't think I would have ever found that. Nice detective work!

rafmag, 2023-01-20:
Do you think it might be worth it to implement ASCII85 encoding? If so, I can work on it. Otherwise we can leave it at HEX Encoding.

StefanSchippers, 2023-01-20:
i gave a look. the problem is not the encoder, for me it is how to embed
the encoded data into a postscript. it can not be embedded as a (....) text
object and the size reduction is (5/4) / 2.
On the other hand specifying jpeg_quality=30 as attribute instead of
default 100 will reduce jpeg data 6 - 7x with negligible degradation on
many  examples i have tested...
If you think it is easy go for it, but what you have done so far is already
great...

rafmag, 2023-01-23:
I went for it. Let me know what you think. I never really liked the Hex encoding. + the PS formatting around the images is more readable now.

rafmag, 2023-01-23:
I just noticed that I forgot to free `*paddedData`  at the end of the `ascii85_encode` routine. Can you please add!

rafmag, 2023-01-23:
What are these id numbers in calls like:
`my_malloc(1662, *output_length +1);`

StefanSchippers, 2023-01-23:
each libc call that allocates/frees memory (alloc, realloc, strdup, free, ...) is wrapped into a function ( my_*() )  and has a unique ID. 
If I run xschem in debug mode:
xschem -d 3 -l log
all allocations / reallocations / frees are written to log.

In the src/ directory there is a track_memory.awk script:

if I run xschem with above debug command and for example do a pdf export
i can later do (in src/):
 ./track_memory.awk ~/log

I get this output if the last paddedData is not freed:

```
schippes@asus:~/xschem-repo/trunk/src$ ./track_memory.awk ~/log
peak allocated memory = 136023913
Total allocated memory after exit = 165356
# of malloc's = 6899
# of realloc's = 22887
# of free's = 23751
  address[ 0x55a685b2f930, 1469 ]= 3616
egrep -n 'my_(malloc|calloc|realloc|free|strdup|strdup2)\(1469,' *.c xschem.h
    save.c:307:  unsigned char *paddedData = my_calloc(1469, input_length+padding, 1);
  address[ 0x55a685b30760, 1469 ]= 3832
egrep -n 'my_(malloc|calloc|realloc|free|strdup|strdup2)\(1469,' *.c xschem.h
    save.c:307:  unsigned char *paddedData = my_calloc(1469, input_length+padding, 1);
  address[ 0x55a685b265c0, 1469 ]= 31864
egrep -n 'my_(malloc|calloc|realloc|free|strdup|strdup2)\(1469,' *.c xschem.h
    save.c:307:  unsigned char *paddedData = my_calloc(1469, input_length+padding, 1);
  address[ 0x55a685ba51b0, 1469 ]= 56028
egrep -n 'my_(malloc|calloc|realloc|free|strdup|strdup2)\(1469,' *.c xschem.h
    save.c:307:  unsigned char *paddedData = my_calloc(1469, input_length+padding, 1);
  address[ 0x55a67db13490, 1469 ]= 35620
egrep -n 'my_(malloc|calloc|realloc|free|strdup|strdup2)\(1469,' *.c xschem.h
    save.c:307:  unsigned char *paddedData = my_calloc(1469, input_length+padding, 1);
  address[ 0x55a685b490c0, 1469 ]= 34396
egrep -n 'my_(malloc|calloc|realloc|free|strdup|strdup2)\(1469,' *.c xschem.h
    save.c:307:  unsigned char *paddedData = my_calloc(1469, input_length+padding, 1);
Number of unfreed pointers = 6
Total leaked memory = 165356
```

so it is telling me there is a memory leak. the 1469 number tells you the offending malloc/strdup/realloc instruction, which is used to grep the line from the sources.
Freeing the paddedData in ascii85_encode fixed the issue:

```
schippes@asus:~/xschem-repo/trunk/src$ ./track_memory.awk ~/log
peak allocated memory = 135893045
Total allocated memory after exit = 0
# of malloc's = 6956
# of realloc's = 23530
# of free's = 24329
Number of unfreed pointers = 0
Total leaked memory = 0
```

rafmag, 2023-01-23:
Very clever. I might "borrow" this idea :)

StefanSchippers, 2023-01-23:
A nice addition would be to get rid of adding these IDs, may be some script / macro doing this automatically. I have a semi-automated approach. It was fine when xschem was small, may be it is a bit tedious now. Will look for a better and automatic implementation. However the embedded memory checker helped me a lot to find memory leaks.

StefanSchippers, 2023-01-23:
Another useful thing was to wrap the `free(ptr)` as `my_free(id, &ptr)`. this will free the pointer ptr points to and set ptr to NULL. If I use a freed pointer by mistake it will bomb the program immediately instead of doing unpredictable things that could get unnoticed abd bomb on the field later.

rafmag, 2023-01-23:
Since I opened this issue, and I feel it's sufficiently resolved now, I'm closing it. Feel free to re-open. Thanks!

closed 2023-01-23 by rafmag

StefanSchippers, 2023-01-23:
the size difference is significant as expected:
```
-rw-r--r--   1 schippes schippes  392947 Jan 23 12:37 poweramp_hexbin.ps
-rw-r--r--   1 schippes schippes  273637 Jan 23 12:37 poweramp_ascii85.ps
```
the interesting fact is that if pdf export is wanted the ghostscript translation (ps2pdf) 
produces the same size pdf, although the two files are not byte to byte identical (very few bytes are different, may be timesamps or something...) 
```
-rw-r--r--   1 schippes schippes  185011 Jan 23 12:48 poweramp_hexbin.pdf
-rw-r--r--   1 schippes schippes  185011 Jan 23 12:48 poweramp_ascii85.pdf
```
Thank you again for this great work