StefanSchippers/xschem #63: Plotting differential voltages...

rafmag, 2022-08-01:
How to you probe a differential voltage (the difference in voltage between 2 nets)? I can only send 1 net to BeSpice (by using Alt-G) and there is no way to edit the trace after.

StefanSchippers, 2022-08-02:
Ngspice does not save differences of node voltages. It leaves this task to the wave viewer. 
Ngspice internal plot command is able to plot this difference, like in:
plot v(net1, net2)
I don't know if gaw or BeSpice are able to do math on saved vectors. I believe BeSpice requires the paid license for this.
Anyway xschem internal wave viewer is able to do that. Attached picture shows a differential voltage plotted. 
Xschem uses RPN postfix notation for math expressions on waves, the format for a voltage difference is:
`"Vdiff; net1 net2 -"`
Above expression (in double quotes) means: 
Create a new waveform that is named Vdiff and each point is calculated as net1 - net2
Another example:
`"r_power; inx vss - i(@r7[i]) * 1e-3 ravg()"`
means: take the voltage difference across a resistor (inx - vss) , multiply by current on R7 (i(@r7[i]), take a 1msec running average to smooth the curve, showing the power dissipated in resistor.
[image]

rafmag, 2022-08-03:
Since differential voltage measurement is fairly common, I was hoping for a diff probe. The commercial tool I use does it like this:

1) you select a net.
2) press a "Set as reference" button
3) when you probe other nets, it is plotted with respect to the reference net.
4) if you press the "set as reference" button with no net selected, the reference goes back to gnd. and probing goes back to normal.

Would it be possible to implement a similar feature? I suspect there would be interest from others in something like that. I would definitely find it very useful. 

Thoughts?

P.S. I checked with Gerard (BeSpice Wave) He gave me the TCL commands to plot differential voltages, but you were right, it's only part of the paid version.

StefanSchippers, 2022-08-06:
I think this feature is potentially doable with a tcl procedure. 
some commands are available to query / update object attributes:
 set xx [xschem getprop rect 2 **1** **node**]
 puts in variable 'xx' the node attribute (list of plotted nodes) of graph number 1 in schematic.
 This variable can be updated:
 append xx "\n\\\"outp outm -\\\""
  This adds another line with "outp outm -" (outp voltage referenced to outm)
  the updated node attribute can be written back:
  xschem setprop rect 2 1 node $xx
  and you will see the additional waveform in graph.

StefanSchippers, 2022-08-09:
> The commercial tool I use does it like this:
>
>     1. you select a net.
>
>     2. press a "Set as reference" button
>
>     3. when you probe other nets, it is plotted with respect to the reference [...]

Regarding above procedure, do differential waves appear differently? does the user notice it is a differential signal? 
I am asking to avoid situations where user specified a reference net and later looks at the differential wave thinking it is referenced to ground.

rafmag, 2022-08-09:
They do appear different in the way that the are labeled:
[image]
The signal on the left is referenced to gnd, the signal on the right is differential.

StefanSchippers, 2022-08-09:
Thank you, yes, this looks the way to go. Either print V(a)-V(b)  (or a-b, shortcut only for voltages as i usually do) or V(a,b) taking the spice syntax.

StefanSchippers, 2022-08-09:
Funny fact I created a dff probe symbol some time ago (`devices/spice_probe_vdiff.sym`) , that connects to two nets and prints a .save V(net1,net2). Only later i realized that ngspice does never save voltage differences, it just saves net1 and net2, leaving the math to the wave viewer.

StefanSchippers, 2023-06-07:
Lets do as ngspice embedded plot function: 
xschem can calculate expressions in its wave window:
`"Vdiff; V(A) V(B) -"`
will display a waveform Vdiff that is V(A) - V(B). xschem uses RPN algebraic notation (easier to code).