StefanSchippers/xschem #64: Bug when dividing voltages in an AC sweep

rafmag, 2022-08-03:
I noticed that when I try to divide 2 voltages in an AC sweep using the following (example):
`CMRR; V(ADM) V(ACM) /` 
I get the wrong result. Looks like the complex numbers aren't being handled correctly. My guess from what I'm seeing is that the voltages are being converted to 'dB' first and then divided.

StefanSchippers, 2022-08-03:
Thank you, will look into this. 
For the time being can't you just do "CMRR; adm acm -" ?

rafmag, 2022-08-03:
Yes. I didn't realize that Xschem was already calculating dB when you used the V(ADM) variable. This is different from standard I think. BeSpice, and the other waveform viewing tools I have output the complex number by default so that you can do whatever math is necessary, and then you can choose to convert to dB or Magnitude or whatever else. 

I think it's better (more general purpose)  to operate on the complex numbers, and then convert with something like `CMRR_DB; V(ADM) V(ACM) / db20`  or `CBRR_PH; V(ADM) V(ACM) / angle` or `CBRR_MAG; V(ADM) V(ACM) / mag`. You never know what crazy calculation the circuit designer is trying to do. But, it will require you to implement complex number math. Developers choice!

StefanSchippers, 2022-08-03:
I agree, however storing raw complex numbers requires to implement or link in a complex math package, while storing numbers in polar form usually makes things simpler. You can do multiplications and division by operating separately on phase and magnitude with real math. This is not the best choice for generality, but in some cases it is simpler as most designers just want to plot gain and phase.

rafmag, 2022-08-03:
Sounds good I just thought I'd put my two cents in :) 

I'll close.

closed 2022-08-03 by rafmag

StefanSchippers, 2022-08-03:
What i probably want to do is to just store the magnitude instead of the dB20 value, and provide a db20() function to calculate magnitude value in dB. This is probably the best compromise. Store plain magnitude and phase, instead of Re and Im parts.

rafmag, 2022-08-03:
Yes, that's a nice compromise. It would help simplify the math.

StefanSchippers, 2022-08-03:
I have done this change. In case of AC simulations xschem computes and stores magnitude and phase for each vector and frequency point. If you have a "vout" variable and need a dB plot the expression to use is:
`"vout [dB]; vout db20()"`
Text before and including ; is optional (sets the label, otherwise use the expresion as label).
This is an incompatible change: older AC plots need to replace each `xxxx` with `"xxxx log20()"` or `"label; xxxx log20()"`