StefanSchippers/xschem #65: TCL Commands to BeSpice Wave

rafmag, 2022-08-03:
I hope I'm not annoying you with all these questions/requests. Feel free to let me know. I'm just excited to get things working just right for our application space.

Is it possible to use a launcher component to send Generic TCL commands to BeSpice Wave?

Or if we wanted to keep it even more general would it be possible to use it to sent TCL commands to other tools if we know the port?

StefanSchippers, 2022-08-04:
For every tool xschem connects to there is a related tcl command. Some tools operate as tcp clients (bespice) and for them xschem becomes a tcp server listening for a specific port. For some other tools (gaw) that behave as tcp servers xschem connects as a client.
For BeSpice, you can send commands to the viewer this way:

`puts $bespice_server_getdata(sock) {set_curve_style "poweramp.raw" "v(outp)" "solid_line" "no symbol" 1 0 255 0}`
`puts $bespice_server_getdata(sock) {add_curve_to_plot "poweramp.raw"  "v(outp)" ""}`
Above example sets a green color for next trace and adds v(outp)
The `$bespice_server_getdata(sock)` is the connection socket file descriptor the string after will be sent to.

rafmag, 2022-08-04:
That works. Thanks!

closed 2022-08-04 by rafmag