StefanSchippers/xschem #66: Importing netlist and CDL format

vietnh89, 2022-08-03:


StefanSchippers, 2022-08-03:
Hi, 
1) People are using xschem netlists to do LVS (Layout Vs Schematic) checks. Xschem has a couple of options in Simulation menu to handle netlists for LVS: 
  1.1) keep the top circuit wrapped into a .subckt definition
  1.2) Do not use spiceprefix (normally 'X') for primitive devices that have a .subckt spice  description, so if a MOS for spice simulation is netlisted as `XM1 drain gate source body sky130_fd_pr__nfet_01v8 W=... L=...`, the same device for LVS will appear in netlist as: `M1 drain gate source body sky130_fd_pr__nfet_01v8 W=... L=...,

2) Some work has been done to parse a spice netlist and create a circuit from it. See the bottom of the README.md [here](https://github.com/StefanSchippers/xschem_sky130/tree/main/xschem_verilog_import).

StefanSchippers, 2022-08-17:
Closing for now as i got no further feedback.

closed 2022-08-17 by StefanSchippers