StefanSchippers/xschem #67: The plots appending [mag]

rafmag, 2022-08-05:
I was making the changes to the AC plots and noticed that it would append [mag] see below. Is there any way to remove this or change the unit?

[image]

StefanSchippers, 2022-08-05:
I have made the fix. Do a `git pull && make install`. The 'Mag' string was previously 'dB' to inform user xschem was storing the value in dB. Since we have now made it more general (just store magnitude and phase) i replaced 'dB' with 'Mag(nitude)', however I agree with you. Since you can define your own alias name there is no need to clutter the wave labels with that info.
Please test and close if all ok

StefanSchippers, 2022-08-05:
PS:  As you probably see the X log axis is now a user option (not implied by AC analysis as before) . There is also a Y log option, the button is there but still not implemented.

rafmag, 2022-08-06:
clean! It's working.

closed 2022-08-06 by rafmag