StefanSchippers/xschem #69: File name issue when using XYCE

rafmag, 2022-08-09:
When using XYCE the default output file name is `schematicName.spice.raw` instead of `schematicName.raw` this has mostly not been an issue since I mostly just changed the TCL command to load data in to graphs to:
`tclcommand="xschem raw_read $netlist_dir/[file tail [file rootname [xschem get current_name]]].spice.raw"`
and have not been using the menu option.

But I just noticed that the cross-probing to BESPICE is then broken. I suspect it's because it's looking for the default name?

Could we add an option to change the output file naming convention when using XYCE? or make it general so the user can specify?

StefanSchippers, 2022-08-09:
Xschem launches Xyce (for example on circuit poweramp_xyce.sch) this way:
Xyce /home/username/.xschem/simulations/poweramp_xyce.sch -r /home/username/.xschem/simulations/poweramp_xyce.raw

If you open Simulations->Configure simulators and tools you see the command:
[image]

The $n (use the Help button in the dialog box for variables explanation) expands to the full path of netlist without extension, and .raw is appended.

rafmag, 2022-08-09:
The issue is that the XYCE output for a print statement goes to a different file than the main output and you can't control the file name from this launch command. The default is the input file name with and appended ".raw".

the is another workaround if I was able to use that variable within the print statement in a netlist statement like this:

`format=".PRINT @SIMTYPE FORMAT=RAW FILE=$n.raw @value` 

but I don't know how to get the $n.raw in there.

StefanSchippers, 2022-08-09:
Can you try this?
`format=".PRINT @SIMTYPE FORMAT=RAW FILE=@symname\\.raw @value`

StefanSchippers, 2022-08-09:
BTW if a raw file is specified globally on command line and a .print statement goes in a different file if no FILE is specified it looks a bug to me...

rafmag, 2022-08-09:
:) you try telling them that! I work with the XYCE team and we discussed this. they don't see it as a bug. It's some sort of HSPICE standard they are following. They think the default should be the full schematic name with the extension included and the ".raw" appended. They bounced the ball back to your court on this. Please help :)

rafmag, 2022-08-09:
> Can you try this?
> `format=".PRINT @SIMTYPE FORMAT=RAW FILE=@symname\\.raw @value`

symname got replaced with the print symbol name. I need the top schematic name.

StefanSchippers, 2022-08-09:
Oh, yes you are right.
You can do something like this: 

`format="tcleval(.PRINT @SIMTYPE FORMAT=RAW FILE=[file tail [ file rootname [xschem get schname 0] ] ].raw )"`

tcleval() will pass the expression to the tcl interpreter.

`xschem get schname 0`  will return the name of the toplevel (level=0) schematic.
The point here is that if you have a schematic with multiple hierarchy levels a probe symbol in a lower hierarchy will probably need to print into the top level filename.

StefanSchippers, 2022-08-09:
May be i can add a @topschname that does the job, similar to predefined @symname @pinlist and other attributes. This saves the tcleval() step which is the slow path.

StefanSchippers, 2022-08-09:
> :) you try telling them that! I work with the XYCE team and we discussed this. they don't see it as a bug. It's some sort of HSPICE standard they are following. They think the default should be the [...]

I believe they have their reasons to do so. Other simulator devs will choose another style. There is no way to have a uniform convention that fits all. Fortunately there are commands / options to set the filename so I think i will enforce one naming convention  to all... Until a new simulator comes out that does not allow output file specifications and we will be fucked.

StefanSchippers, 2022-08-09:
Ok I have added `@topschname` predefined attribute that expands to the toplevel schematic (without .sch extension):

format reduces to:
`format=".PRINT @SIMTYPE FORMAT=RAW FILE=@topschname\\.raw @value"`

rafmag, 2022-08-10:
actually I ended up using `xschem get current_name` since top for me was more of a display only schematic. if you could also make a "current schematic" attribute, that would be helpful.

Either way I think the issue is solved now, so closing. Thanks!

closed 2022-08-10 by rafmag

StefanSchippers, 2022-08-10:
there are 2 attributes now:
`@schname_ext`  --> does the same as TCL `xschem get current_name`
`@schname` --> same as above but with no extension (no ending .sch)