StefanSchippers/xschem #70: Problem with array devices

rafmag, 2022-08-09:
I have a transistor symbol defined like this:
[image]

I then instantiate it like this:
[image]

When I create the netlist only the first device has the designator. It looks like this:

`XQ1[0] net1 net2 net3 modelname`
`Q1[1] net1 net2 net3 modelname`
`Q1[2] net1 net2 net3 modelname`

Is this a bug, or am I doing something wrong?

StefanSchippers, 2022-08-09:
This is a known issue and has to do with the netlist finishing (post processing) done by an awk script.
Xschem spits out a netlist in an intermediate form, doing the hard work (resolving connections, reporting errors etc) 
The final spice netlist is generated with an awk script. Xschem is written in C and so very fast. The logic representation of the schematic is extracted and a netlist is generated. Awk is very powerful on text processing (and also fast, faster than perl/python and tcl itself) so the syntax finishing is done by an awk script. This is good since generating a spice netlist dialect for another spice simulator should in many cases not require to change the netlister engine.
In your case the intermediate form is generated as follows:
`XQ1[0],Q1[1],Q1[2] ?1 Z ?1 A  modelname`
The Q1[0:2] is expanded and some information on port multiplicity is printed (?1 in this case)
So you see why the postprocessor names the 1st device as XQ1[0] and does not add the prefix to following instances.

The solution is to define the format attribute this way:
`format="#@designator\\#@name @pinlist @model"`

In this case the intermediate form will look like: 
`#X#Q1[0],Q1[1],Q1[2] ?1 Z ?1 A  modelname`
and the postprocessor knows to add the prefix to all instances.

The double slash in the format string tells xschem that 'designator' is a variable to be substituted and not 'designator#'
Usually if you have a variable not separated by spaces you must escape the character that is not part of the variable name, like `@designator\#`, however since we are inside double quotes we must use a double \\\\ to enter a \\ inside the quoted format string.

rafmag, 2022-08-09:
wow. ok, that was a little hard to follow, but your solution worked, so I'm happy :)

closed 2022-08-09 by rafmag